BatchDrake/SigDigger #4: "Failed to start capture due to errors:"

miek, 2019-08-17:
When I start a capture, I get a popup with just the text "Failed to start capture due to errors:". I've tried with both a HackRF source and a file source (complex f32 samples) and get the same result.

I'm running Ubuntu 18.04 and I am running a slightly older Qt version than suggested (5.9.5) but I think it's unrelated?

I'm happy to dig in and do some more debugging but I don't really know where to start, so I'd need some advice on things to try.

BatchDrake, 2019-08-17:
Hi miek,

Is there any additional information in that popup message? Also, a file named `~/.suscan/config/uiconfig.xml` should have been created in your system, containing SigDigger's config (including the signal source config). Could you attach this file here? 

Thank you,

miek, 2019-08-17:
There's no other information in the popup.

I've attached uiconfig.xml for each configuration (renamed 'cause github is scared of xml files apparently...).

[uiconfig-filesource.txt](https://github.com/BatchDrake/SigDigger/files/3512124/uiconfig-filesource.txt)
[uiconfig-hackrf.txt](https://github.com/BatchDrake/SigDigger/files/3512125/uiconfig-hackrf.txt)

Thanks

BatchDrake, 2019-08-17:
I've just installed Ubuntu 18.04 and reproduced this behavior, I'm already looking into it. The cause of this issue seems to be somewhere in the constructor of the suscan's analyzer object, file `analyzer/analyzer.c`, right after line 1364, in case you want to start debugging.

BatchDrake, 2019-08-17:
Well, it ended up being in `suscan` :smile: 

File `analyzer/analyzer.c`, function 1232:

```c
SUBOOL
suscan_analyzer_set_iq_reverse(suscan_analyzer_t *analyzer, SUBOOL rev)
{
  analyzer->iq_rev = rev;
}
```

Must be modified to:

```c
SUBOOL
suscan_analyzer_set_iq_reverse(suscan_analyzer_t *analyzer, SUBOOL rev)
{
  analyzer->iq_rev = rev;
  return SU_TRUE;
}
```

I'm adding a hotfix right now, could you try to change this file and verify whether it works?

miek, 2019-08-17:
Yep, that works. Thanks very much for solving that!

closed 2019-08-17 by miek

BatchDrake, 2019-08-17:
Changes added to ffdcc8cfb4ca73524859733b6498ab0870ea8fd8, I'm merging back to master. Thank you for your feedback!