BatchDrake/SigDigger #5: Error in InspectorMessage

andruxa-smirnov, 2019-08-17:
Ubuntu 18.04, x64
Qt 5.12
Suscan/Messages/InspectorMessage.cpp: In member function ‘unsigned int Suscan::InspectorMessage::getBasebandRate() const’:
Suscan/Messages/InspectorMessage.cpp:84:25: error: ‘struct suscan_analyzer_inspector_msg’ has no member named ‘fs’
   return this->message->fs;
                         ^~
Suscan/Messages/InspectorMessage.cpp: In member function ‘float Suscan::InspectorMessage::getEquivSampleRate() const’:
Suscan/Messages/InspectorMessage.cpp:90:25: error: ‘struct suscan_analyzer_inspector_msg’ has no member named ‘equiv_fs’
   return this->message->equiv_fs;
                         ^~~~~~~~
Suscan/Messages/InspectorMessage.cpp: In member function ‘float Suscan::InspectorMessage::getBandwidth() const’:
Suscan/Messages/InspectorMessage.cpp:96:25: error: ‘struct suscan_analyzer_inspector_msg’ has no member named ‘bandwidth’
   return this->message->bandwidth;
                         ^~~~~~~~~
Suscan/Messages/InspectorMessage.cpp: In member function ‘float Suscan::InspectorMessage::getLo() const’:
Suscan/Messages/InspectorMessage.cpp:102:25: error: ‘struct suscan_analyzer_inspector_msg’ has no member named ‘lo’
   return this->message->lo;
                         ^~
Makefile:1660: recipe for target 'InspectorMessage.o' failed
make: *** [InspectorMessage.o] Error 1

BatchDrake, 2019-08-17:
I added extra fields to the inspector message object these days in Suscan in order to support dynamic channel adjustment from SigDigger. Just go to suscan's source tree, run:

```
% git pull
% make
% sudo make install
```

And try to compile SigDigger again.

andruxa-smirnov, 2019-08-17:
Thanks. It's work. I was build all parts and start testing

closed 2019-08-17 by andruxa-smirnov