BatchDrake/SigDigger #6: Crash into analyze file

andruxa-smirnov, 2019-08-17:
When i start Open Inspector for IQ-rec file, app crash with messages:
Gtk-Message: 22:38:03.538: GtkDialog mapped without a transient parent. This is discouraged.
"Spectrum zoom: 1.1x"
"Spectrum zoom: 1.2x"
"Spectrum zoom: 1.4x"
"Spectrum zoom: 1.5x"
"Spectrum zoom: 1.7x"
"Spectrum zoom: 1.9x"
"Spectrum zoom: 2.1x"
"Spectrum zoom: 2.2x"
"Spectrum zoom: 2.5x"
"Spectrum zoom: 2.7x"
"Spectrum zoom: 3.0x"
"Spectrum zoom: 3.3x"
"Spectrum zoom: 3.7x"
"Spectrum zoom: 4.1x"
"Spectrum zoom: 4.4x"
"Spectrum zoom: 4.9x"
"Spectrum zoom: 5.5x"
"Spectrum zoom: 5.6x"
"Spectrum zoom: 5.1x"
"Spectrum zoom: 4.6x"
"Spectrum zoom: 3.6x"
double free or corruption (fasttop)double free or corruption (fasttop)double free or corruption (fasttop)

Thats happen not always (memory leaking, may be)

BatchDrake, 2019-08-17:
> Gonzalo José Carracedo Carballal

Interesting. I've been trying to track invalid memory accesses and frees
before releasing this, but I certainly missed something. Could you provide
some additional information on how to reproduce this?

Cheers,

andruxa-smirnov, 2019-08-18:
Yes, from time to time the error is reproduced. Log:

$ /opt/SigDigger/bin/SigDigger
[INFO] bladerf_open_with_devinfo()
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":127] FPGA version (v0.10.2) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO] bladerf_get_serial() = dd653505c91bb931d09fffec160ccbf4
[INFO] setSampleRate(1, 1.000000 MHz), actual = 1.000000 MHz
[INFO] setSampleRate(0, 1.000000 MHz), actual = 1.000000 MHz
[INFO] bladerf_close()
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
[INFO] bladerf_open_with_devinfo()
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":127] FPGA version (v0.10.2) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO] bladerf_get_serial() = dd653505c91bb931d09fffec160ccbf4
[INFO] setSampleRate(1, 1.000000 MHz), actual = 1.000000 MHz
[INFO] setSampleRate(0, 1.000000 MHz), actual = 1.000000 MHz
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/bladerf.c":795] Clamping bandwidth to 1500000Hz
[INFO] setSampleRate(1, 2.000000 MHz), actual = 2.000000 MHz
00Gtk-Message: 09:58:38.717: GtkDialog mapped without a transient parent. This is discouraged.
[INFO] bladerf_close()
Gtk-Message: 09:59:16.529: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 09:59:53.146: GtkDialog mapped without a transient parent. This is discouraged.
[INFO] bladerf_open_with_devinfo()
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/version_compat.c":127] FPGA version (v0.10.2) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO] bladerf_get_serial() = dd653505c91bb931d09fffec160ccbf4
[INFO] setSampleRate(1, 1.000000 MHz), actual = 1.000000 MHz
[INFO] setSampleRate(0, 1.000000 MHz), actual = 1.000000 MHz
[INFO @ "/build/bladerf-zjSF8E/bladerf-2017.12-rc1/host/libraries/libbladeRF/src/bladerf.c":795] Clamping bandwidth to 1500000Hz
[INFO] setSampleRate(1, 2.000000 MHz), actual = 2.000000 MHz
00[INFO] bladerf_close()
Gtk-Message: 10:01:00.479: GtkDialog mapped without a transient parent. This is discouraged.
Ошибка сегментирования (стек памяти сброшен на диск)

Steps to reproduce:
1. Set bladerf sdr as source.
2. Run.
3. Record bandwidth to file.
4. Stop record, stop sdr.
5. Set recordedfile as source.
6. Run.
7. Open inspector (PSK mode).
8. Crash.

BatchDrake, 2019-08-19:
I'm afraid I'm failing to reproduce (also I don't have any bladeRF at reach now). Would you mind doing the following?

1. Run ```gdb /opt/SigDigger/bin/SigDigger```
2. In the ```(gdb)``` prompt, type ```run``` and press Enter
3. SigDigger's window should show up. Try to reproduce until, instead of closing, it gets unresponsive
4. The terminal window should say that the program received a signal (probably SIGSEGV)
5. In the ```(gdb)``` prompt, type ```bt``` and press Enter
6. A backtrace should appear. Copy and paste it here.

andruxa-smirnov, 2019-08-20:
#0  0x00007ffff56ead6e in __pthread_barrier_wait (barrier=0xff193b3e000002b8) at pthread_barrier_wait.c:110
#1  0x00007ffff777437c in suscan_analyzer_source_barrier (analyzer=<optimized out>) at analyzer.c:781
#2  0x00007ffff777e2cc in suscan_inpsched_barrier_cb (mq_out=<optimized out>, wk_private=<optimized out>, cb_private=<optimized out>) at inspsched.c:88
#3  0x00007ffff7786f49 in suscan_worker_thread (data=0x555555e8ce90) at worker.c:99
#4  0x00007ffff56e36db in start_thread (arg=0x7fffd9152700) at pthread_create.c:463
#5  0x00007ffff4acd88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

BatchDrake, 2019-08-21:
Confirmed, I was able to replicate in exactly in the same point. Already looking into it.