BatchDrake/SigDigger #7: How add rl-sdr device?

andruxa-smirnov, 2019-08-18:
rtl-sdr installed, but in source list of sdr devices not present. 
Log of rtl_test:
$ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0013 tuner
Supported gain values (23): -9.9 -7.3 -6.5 -6.3 -6.0 -5.8 -5.4 5.8 6.1 6.3 6.5 6.7 6.8 7.0 7.1 17.9 18.1 18.2 18.4 18.6 18.8 19.1 19.7 
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...

nmaster2042, 2019-08-18:
Hi. Seems it needs the SoapySDR + SoapyRTLSDR.

SoapySDR is an abstraction layer that provide support for various SDR hardware.

See here:
https://github.com/pothosware/SoapySDR
https://github.com/pothosware/SoapyRTLSDR

Hope it will help.

BatchDrake, 2019-08-18:
Yep, in the end, SigDigger just wraps around SoapySDR in order to detect SDR devices. Installing Soapy's rtl-sdr module should make it work (you shouldn't need to recompile in order to do this, it would be enough by installing the missing modules).

ganlubbq, 2019-08-18:
> Yep, in the end, SigDigger just wraps around SoapySDR in order to detect SDR devices. Installing Soapy's rtl-sdr module should make it work (you shouldn't need to recompile in order to do this, it would be enough by installing the missing modules).

How to install the missing modules?

BatchDrake, 2019-08-18:
@ganlubbq it depends on how you've installed SoapySDR in your system and the specific distribution you are using. In Debian and recent Ubuntu releases there should be a package named something like ```soapysdr-module-all``` (the exact name may be different) containing all the modules you may need.

andruxa-smirnov, 2019-08-18:
Hm. Soapy (self and modules) was installed, but no rtlsdr modules appeared

andruxa-smirnov, 2019-08-18:
After recompiling suscan i received next messages:
$ suscan
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
[INFO] bladerf_open_with_devinfo()
[ERROR] bladerf_open_with_devinfo() returned -7 - No device(s) available
suscan: suscan library loaded successfully.
---------8<-------------------------------------
(e) exception in "sdev = SoapySDRDevice_make(dev->args)" (source.c:189)
(e) exception in "suscan_source_device_populate_info(device)" (source.c:1300)
(!) Failed to populate device info
(e) exception in "sdev = SoapySDRDevice_make(dev->args)" (source.c:189)
(e) exception in "suscan_source_device_populate_info(device)" (source.c:1331)
(!) Failed to populate device info

Any ideas?

BatchDrake, 2019-08-18:
These messages are simply saying that, although the bladeRF module is
present, it cannot be used as source because it is not detecting any
bladeRF device.

Unless you've plugged a bladeRF device, this is not an error.

pborges, 2019-08-20:
I also have this problem, I have SoapySDR installed on Ubuntu 18.04
My RTL2832U is discovered however I only see "Dummy Device" in the AppImage

`SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: pcm device (hw:1,0) data format not supported by RtAudio.

RtApiAlsa::getDeviceInfo: pcm device (hw:1,1) data format not supported by RtAudio.

Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = hw:HDA Intel PCH,0

Found device 1
  default_input = False
  default_output = False
  device_id = 6
  driver = audio
  label = 

Found device 2
  default_input = False
  default_output = False
  device_id = 9
  driver = audio
  label = hw:USB Audio Device,0

Found device 3
  default_input = False
  default_output = False
  device_id = 10
  driver = audio
  label = default

Found device 4
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
`

BatchDrake, 2019-08-20:
Mmh, I thought I fixed it in the last release, but maybe I'm still missing
something. Tests in my Ubuntu 16.04 (release 0.0.5) worked well, both with
RTL-SDR and AirSpy. What AppImage release are you using?

pborges, 2019-08-20:
I believe it was 0.0.4-alpha I will try 0.0.5-alpha tonight and report back. Thank you.

BatchDrake, 2019-08-20:
Since debugging AppImage problems is starting to become a little
cumbersome, I'm going to open a feature request to add a debug option to
check whether modules are being loaded right. I'll keep you all updated.

pborges, 2019-08-21:
0.0.5 is working for me thank you

closed 2023-04-12 by BatchDrake