BatchDrake/SigDigger #8: SigDigger appimage audio clipping at 32,000 up

2WR3505, 2019-08-19:
Wavin drake, i was testing this on my skywave linux install, and the audio gets choppy at 32,000 and up. is there any thing i can do to fix this by tweakin settings in the program?

BatchDrake, 2019-08-19:
I suspect what is going on. Can you provide details on signal source, audio
cutoff frequency, and modulation?

2WR3505, 2019-08-20:
signal source is truckers on 11 meters AM, cutoff is 22,050 KHz. i noticed it stops clipping after awhile.

the audio remains horrible though

BatchDrake, 2019-08-21:
Sure, but I mean, what SDR device are you using? What is the capture sample rate?

> the audio remains horrible though

Most (if not all) AM stations transmit at a cutoff frequency of 4 kHz. At bigger cutoff frequencies (i.e. 22 kHz like in your case) you start to get out-of-band noise coming from nearby stations. Have you tried to adjust it?

Also, I fixed an issue related to sample rate yesterday (some SDR devices will silently capture at a different rate if you specify an unsupported rate), but I haven't merged to master yet. This problem caused a lot of trouble regarding audio preview, and can be potentially related to this issue. I'll update you once it's done.

PS: there is a problem regarding audio filter stability at very low frequencies (i.e. below 3 kHz). But in that case the only thing you'd hear is a popping sound. I guess that's not your case, right?

2WR3505, 2019-08-21:
your thinking of the bandwidth of the transmission, which is actually 6 KHz. i have the sample rate the same as i use on GQRX. 22 KHz is related to audio quality. i got the audio quality cutoff as high as it can go, and that doesnt seem to fix it.

i have it set to 6 KHz wide for standard AM bandwidth.

yea, i'll try 0.5.0 of the alpha app image and see if that changes anything.

and yea, thats not my case, lol. i'll have to roll a quick video gate using my kraco mayday 2 :)

BatchDrake, 2019-08-21:
The audio preview is a little counterintuitive (this is totally my fault) and perhaps it's time to rethink the whole thing entirely. Roughly speaking, there are two filters working here:

* The channel filter, which is inserted right before the detector. This is the one you can configure manually in the spectrum view by adjusting the gray box around the center frequency. Although this filter can arbitrarily wide, the audio preview limits silently to 200 kHz (so you can hear WBFM stations)
* The audio cutoff filter. This one is after the detector, and it is aimed to remove nasty things like everything coming after the WBFM pilot tone, or any high frequency interference in the AM signal.

Playing with it for a bit, I've realized that making the channel filter narrower (but keeping it at least 6 kHz wide) distorts the signal somehow. Could you try different combinations of the following?

* Reduce volume to nearly 1% in the Audio preview panel (and raise the volume of your computer so you can actually hear something). Does the quality increase?
* Increase the channel filter bandwidth (try something wider than 200 kHz). Play with the volume again. Any good?

If you don't get any improvements, please try to record some seconds of the audio output to a file (using Audacity, for instance) and send it back to me.

2WR3505, 2019-08-21:
ah, ok, good info :)

I'll try those 2 tips there, but here's a current videogate of where i am at right now :)

https://www.youtube.com/watch?v=f4u7m4hUF68&feature=youtu.be

BatchDrake, 2019-08-21:
This was extremely helpful. Yes, there are two different issues going on here:

* The output gain is too high, and gets clipped by the soundcard.
* The audio cutoff filter works poorly on low frequencies (this is a consequence of using IIR filters instead of longer but more stable FIR filters). 

Since these are two different audio inspector issues in Suscan, I'm moving this issue to Suscan's project page. I hope to have it fixed before the week ends.