BatchDrake/SigDigger #13: Bladerf bandwidth limited to 2 MHz

andruxa-smirnov, 2019-08-22:
When i set bladerf bandwidth more, than 2MHz (more 2000000 samples per second), bandwidth all time limited to 2 MHz.

BatchDrake, 2019-08-22:
Confirmed, working on it.

BatchDrake, 2019-08-22:
I've just commited something to suscan's master branch that may be related to what you are experiencing. I've just checked it out with AirSpy at 6000000 and Bandwidth is limited to 6 MHz (as expected). Rebuild suscan from master and try again. If you still fail to make it work, could you provide a screenshot?

andruxa-smirnov, 2019-08-23:
Rebuilding don't solve problem
[image]

BatchDrake, 2019-08-23:
That's strange, I can increase the bandwidth up to the sample rate. Can you
provide some screenshots of what you're attempting?

andruxa-smirnov, 2019-08-24:
[image]
[image]

BatchDrake, 2019-08-24:
You are not running the latest version of SigDigger (there should be a bandwidth control in the source panel). Pull from master (from SigDigger, not Suscan) and build again. You should be able to adjust the bandwidth dynamically after that.

andruxa-smirnov, 2019-08-24:
What i doing not so?
I deleted SigDigger folder. I clone master branch newly, after that i do "qmake", "make" and "sudo make install", but that don't solve problem.
More one intresting thing - i see bladerf device double in sdr sources list. Why?

BatchDrake, 2019-08-24:
I've just cloned SigDigger from master. After building I do see the bandwidth adjust widget, look:

[image]

Please check again whether you are actually running the right executable. This dynamic bandwidth control feature is already in master.

> More one intresting thing - i see bladerf device double in sdr sources list. Why?

This happens if the device label has changed since the last time you plugged it (or if you have two different BladeRF devices, and the one you plugged last time is different from the one you have plugged now). This is the expected behavior. What is not expected is the BladeRF identification string to change between executions.

closed 2019-08-25 by andruxa-smirnov