BatchDrake/SigDigger #14: USRP Support

kevincroissant, 2019-08-24:
Hi, 

I installed soapysdr-module-uhd and soapysdr-module-all with apt-get on my Debian Buster system, but I cannot get SigDigger to recognize my Ettus USRP N210 device. uhd_find_devices and gqrx have no issue finding it.  I am using the SigDigger 0.0.6 AppImage.
If you don't have access to a USRP for testing, let me know and I will help test things out to help debug.

P.S. I tested it out using an AirSpy, it works great so far!

Thanks!

BatchDrake, 2019-08-24:
I didn't add USRP support because I thought it may make the AppImage a
little too big, but I wouldn't mind adding the missing modules in the next
release.

PS: I only have an RTL-SDR, an AirSpy and a HackRF, so any feedback about
other devices will always be welcome. Could you try to build SigDigger from
sources and tell me if it works with your USRP?

Thanks!

kevincroissant, 2019-08-25:
After building from source, I was able to connect to the USRP. Here's what I've noticed so far:
1. When first starting up, there is a massive DC offset, goes to 0dB. The DC remove button works and is absolutely necessary. This doesn't show up in GQRX.
2. I cannot modify "ADC-Fine" gain. It's stuck at 0. My PGA0 gain is also maxing out at 31dB, vs 31.5dB in gqrx.
3. It seems like the noise floor is much higher in SigDigger for some reason, when compared against gqrx. I attached screenshots of the waterfall+FFT and configs for each program:

Gqrx:
[image]

Gqrx config:
[image]
[image]

SigDigger:
[image]

SigDigger config:
[image]
[image]

I also just noticed that I can't adjust the FPS rate of the FFT in SigDigger, is that a bug or an upcoming feature?

I'll let you know if I find anything else.

BatchDrake, 2019-08-25:
Wow, this is what I call useful feedback, thank you very much! Since this refers to a bunch of things now, I'll open separate issues tonight. In the meantime:

> 1. When first starting up, there is a massive DC offset, goes to 0dB. The DC remove button works and is absolutely necessary. This doesn't show up in GQRX. 

So you suggest having the DC Remove button enabled by default, right? I wasn't sure whether this made sense when I first added it.

> 2. I cannot modify "ADC-Fine" gain. It's stuck at 0. My PGA0 gain is also maxing out at 31dB, vs 31.5dB in gqrx.

This means that my assumption that device gains are integers was wrong. What is the maximum value for ADC-fine, according to Gqrx?

> 3. It seems like the noise floor is much higher in SigDigger for some reason, when compared against gqrx. 

This is interesting. I didn't compare the spectrums of Gqrx and SigDigger too much. Do you think this may be related to the ADC-Fine gain issue?

> I also just noticed that I can't adjust the FPS rate of the FFT in SigDigger, is that a bug or an upcoming feature?

This is an upcoming feature. Since the way Gqrx handles rates is fundamentally different from SigDigger's, I decided to lower the priority of this. The same thing happened to Time span. Is it critical for you to adjust this now? The workaround is to open the config dialog and set the spectrum refresh interval in the Analyzer parameters dialog manually.

kevincroissant, 2019-08-26:
Glad to help!
I think having the DC remove auto-enable is a workaround, but a decent enough one. I'm not aware of a scenario where it is detrimental to use, but it might exist. Like I said, I don't see this DC offset elsewhere. Maybe it's a SoapySDR thing?

ADC-Fine varies from 0-1.0dB, for the Ettus WBX daughterboard.
I'm pretty sure the spectrum issue is unrelated to ADC-Fine, since it's only 1dB of gain.

As for the FFT rate, it would be nice to have but not critical. I found that when I increased the FFT size to 8192 to match what gqrx can handle nicely on my machine, SigDigger was consuming more resources and doing it more often. The workaround is good enough for now.

BatchDrake, 2019-08-27:
That is exactly what happened, since this was configurable in SoapySDR I assumed that automtic DC correction could be troublesome in some scenarios (cheap receivers with poor implementations, maybe)? Dunno.  Anyways, I've just fixed this in Suscan and merged to master.

Regarding the issue with gains (https://github.com/BatchDrake/SigDigger/issues/16), I've just commited some changes to the SigDigger's branch ```feature/real-gains```. Could you try to build from there and adjust the gains again?

I'm also working on the FFT rate / time span feature, it should be ready before the week ends. Then I'll start investigating the mysteriously high noise floor.

madengr, 2021-12-21:
How did you get SigDigger to recognize the N210 as it requires passing an IP address?  I can access it fine with Soapy but see no way of passing the address when launching SigDigger nor suscli.  Can you post your ~/.suscan/config/sources.xml ?

SoapySDRUtil --probe="driver=uhd,addr=192.168.1.12"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=uhd,addr=192.168.1.12
[INFO] [UHD] linux; GNU C++ version 11.2.1 20211203 (Red Hat 11.2.1-7); Boost_107600; UHD_4.2.0.git-209-gf23ab721
[INFO] [x300_impl.cpp:212] [X300] X300 initialization sequence...
[INFO] [X300] X300 initialization sequence...

suscli devices
 ndx Device name                              Driver   Interface Availability 
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     available

closed 2023-04-12 by BatchDrake