BatchDrake/SigDigger #15: Peak Detection button causes segfault

kevincroissant, 2019-08-24:
Hello,

Using an AirSpy R2 on Debian Buster, with the 0.0.6 AppImage, click the Peak Detection button causes an immediate segfault:
```
/home/kevin/Downloads [kevin@carbon] [11:30]
> ./SigDigger-0.0.6-x86_64.AppImage
[INFO] Using format CF32.
/tmp/.mount_SigDigDAkvFX/AppRun: line 4: 15641 Segmentation fault      $APPDIR/usr/bin/SigDigger.bin
```
Not sure if this is supposed to work yet - if so, let me know how I can help test or what other info you would like.

Thanks!

kevincroissant, 2019-08-24:
I forgot to mention, the Peak Hold button causes the exact same behavior.

BatchDrake, 2019-08-25:
I've been trying to reproduce this behavior -both from sources and AppImage- but no luck so far. Could you attach the XML files saved in ```~/.suscan/config``` to this issue?

Thanks!

kevincroissant, 2019-08-25:
Here you go:
[config.zip](https://github.com/BatchDrake/SigDigger/files/3538302/config.zip)

closed 2023-04-12 by BatchDrake