BatchDrake/SigDigger #20: Question about adding support for new SDRs

mehdideveloper, 2019-10-02:
Hey @BatchDrake 
Is SigDigger fully relying on SoapySDR or you also have to add device-specific configurations to the code?
In a comment to issue #12 you said 

> For AirSpy (and hopefully more radios in the future) there are gain presets that allow you ...

And I wondered what you mean by hopefully more radios in future? Because in another comment you've said

> ... in the end, SigDigger just wraps around SoapySDR in order to detect SDR devices. 

I have access to some other SDRs like different SDRPlays and LimeSDR and I can help in testing, but your comment made me wonder if you have to do some changes in the code. 

Can we define a simple test criteria as: if you can open a device in SigDigger and observe the spectrum/waterfall and also open some inspector tabs without any issues, then the test is successful for that SDR?

BatchDrake, 2019-10-02:
It is currently relying on SoapySDR for all receivers. However, for
specific devices (like AirSpy) I offer something called "gain presets" that
allow you to set many gains at once using a single slider, based on fixed
gain tables that are copied to /usr/local/share/suscan. These tables are
plain XML and hence easy to edit.

Note these tables are simply data files I wrote based on what I know about
the behavior of specific device models, so you don't have to bother setting
the gains manually. They are practical, but not required.

Regarding the test criteria, for now it's okay if you simply see the
waterfall and can browse the spectrum by setting the frecuency while the
radio is running. Channel inspection is absolutely device-independent, so
it will not be necessary to test it explicitly.

Thanks!

mehdideveloper, 2019-10-03:
Ok I tested it with different SDRs and the app worked with everything that has a SoapySDR binding. Here are the devices I tested:

- SDRPlay RSP1A
- SDRPlay RSP2
- RTL-SDR V3
- AirSpy R2
- ADALM-PLUTO (PlutoSDR)
- LimeSDR USB

There's a problem with antenna selection in RSP2 ( choosing "Antenna B"  doesn't change anything and still Antenna A is the main one, but I think it's a SoapySDRPlay problem)

closed 2023-04-12 by BatchDrake