BatchDrake/SigDigger #23: Compile error

BX2ABT, 2019-10-20:
In file included from include/Suscan/Analyzer.h:26,
                 from Suscan/Analyzer.cpp:23:
Suscan/Analyzer.cpp: In member function ‘void Suscan::Analyzer::setFrequency(double)’:
Suscan/Analyzer.cpp:120:59: error: too few arguments to function ‘int suscan_analyzer_set_freq(suscan_analyzer_t*, double, double)’
   SU_ATTEMPT(suscan_analyzer_set_freq(this->instance, freq));
                                                           ^
include/Suscan/Compat.h:29:9: note: in definition of macro ‘SU_ATTEMPT’
   if (!(expr)) {             \
         ^~~~
In file included from /usr/local/include/suscan/analyzer/msg.h:31,
                 from include/Suscan/Message.h:27,
                 from include/Suscan/Analyzer.h:29,
                 from Suscan/Analyzer.cpp:23:
/usr/local/include/suscan/analyzer/analyzer.h:154:8: note: declared here
 SUBOOL suscan_analyzer_set_freq(
        ^~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:1702: Analyzer.o] Error 1

BatchDrake, 2019-10-20:
You seem to be compiling a pretty old version of SigDigger against a relatively recent version of suscan. Clone the latest SigDigger and SuWidgets from master and try again.

BX2ABT, 2019-10-20:
My bad, I guess I mixed up sources. Compiled, installed and running. Cheers!

closed 2019-10-20 by BatchDrake