BatchDrake/SigDigger #48: Error running AppImage : version `ZLIB_1.2.9' not found

LamaBleu, 2020-01-21:
Hi BatchDrake,

First thank you for the fantastic work you are doing on SigDigger.

Trying the AppImage release on my Ubuntu 18.04.3 LTS returns following error :

```
./SigDigger-full-x86_64.AppImage 
/tmp/.mount_SigDigKeelMp/usr/bin/SigDigger.app: /tmp/.mount_SigDigKeelMp/usr/bin/../lib/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)

```
This workaround worked for me: set LD_LIBRARY_PATH variable:
`export LD_LIBRARY_PATH=/lib/x86_64-linux-gnu`

Hope this can help others !
BR

BatchDrake, 2020-01-21:
Wow, nice catch! This is actually a deployment issue that comes from SigDigger relying on the system's libpng instead of the built-in one. I'll fix this ASAP.

BatchDrake, 2020-01-22:
Alright, the laptop with Ubuntu 16.04 that I used for deployment just died. I think this is going to take a while :(

BatchDrake, 2020-01-25:
Hi!

I've just made some tentative changes in the deployment script and came up with a new AppImage file. It turns out that I shouldn't be including libz in the resulting AppImage, and it also lacked a proper library path configuration. Do you mind [trying this one out](http://actinid.org/outcoming/SigDigger-full-x86_64.AppImage) and telling me whether it works, with no need for additional linker flags?

tavi343, 2020-01-31:
The first solution work for me, the last one don't. I have Ubuntu 18.04. I'm not very experienced in this. The compilation of sigutills failed, so this package is great! Thank you.

BatchDrake, 2020-01-31:
Ouch. In any case, do you guys mind sharing the errors you get in the terminal with me? (both build failures or any library / versioning errors). They may help me improve compatibility in future releases.

tavi343, 2020-01-31:
Unfortunately, I don't save the error messages. I will try next week to compile again and save them. For the AppImage the error is the same as in the first post. 
Nice weekend,
Tavi

jslater89, 2020-02-08:
As another data point for you, the most recent release on the releases page didn't work for me (Mint 19.3), but the one you provided in your comment above did.

BatchDrake, 2020-02-08:
I will take all this into account. I'm expecting a new laptop next week, as soon as I have it I will install different VMs to do some intensive testing before dropping a release.

BatchDrake, 2023-04-12:
Hi,

I believe this should be fixed now. I am closing the issue because it is extremely old, feel free to reopen it if the AppImages are still crashing.

Cheers,

closed 2023-04-12 by BatchDrake