BatchDrake/SigDigger #52: Wide bandwidth SDR capture

tavi343, 2020-01-31:
Many SDR on the market have a large instantaneous bandwidth. It will be useful if you can view the full spectrum of the SDR is capable, or maybe more using fast sweeps, to see a very wide bandwidth. Of course, the display will be limited but is not important if you can zoom in at while. At least, the recording should be arbitrary wide bandwidth, in the maximum range of the SDR. For example, there are >100MHz bandwidth SDR like RedPitaya. I would use this for radio astronomy.

BatchDrake, 2020-01-31:
You have just described a feature that is already in the `develop` branch and will be included in the next SigDigger release, named Panoramic Spectrum, although mine covers up to 2 GHz 😁 
[image]
[image]

BatchDrake, 2020-01-31:
Just in case you want to see it in action: https://www.youtube.com/watch?v=IGiGn3_9Sgs&feature=youtu.be

BatchDrake, 2020-01-31:
Duplicate of #32

BatchDrake, 2020-01-31:
Now that I think about it, for radio astronomy purposes, this feature would fall short as its integration time is not well defined (it will average up to 5 samples of the same bin, crunch them into 1 single averaged sample and start again). It could be interesting to add an "integration time" option to fix this. If you are interested, feel free to open a new feature request about this integration time feature.

tavi343, 2020-01-31:
Oho, that's wonderful! For scientific purposes, only the recording should be as wide as the SDR+antenna system can do, as fast as it can, the more the better. A 500MHz bandwidth with 15-20 samples/second will be good enough, of course using a high end SDR like USRP. Then the visualization will be shrink or enlarge at will. I don't know if this "integration time" is important or useful in the recording phase. Is very nice to have a real time waterfall but not important to my application.   
Thank you and have a nice weekend.
Tavi

BatchDrake, 2020-01-31:
Same, thank you for your feedback!

closed 2020-01-31 by BatchDrake