BatchDrake/SigDigger #54: Feature request: Runtime SDR detection

mehdideveloper, 2020-02-04:
It seems right now the list of devices is initialised in the startup. 
It would be a nice feature if we can refresh the list, so if a new device is connected while the app is running, we can detect and start it (without closing and re-opening the app)
For example CubicSDR has a similar button

BatchDrake, 2020-02-04:
You can. Press Ctrl+D.

mehdideveloper, 2020-02-04:
Doesn't work (tried with Ctrl+D and also Cmd+D)

BatchDrake, 2020-02-04:
This is in the ```develop```
[image]
 branch. Try the menu instead (View / Devices)

mehdideveloper, 2020-02-04:
I tried the menu and found another bug: pressed the refresh button and it showed "airspy", while no USB devices are attached to the system! Also `SoapySDRUtil --find` shows nothing (as expected)

BatchDrake, 2020-02-04:
Mmh, does it show it in green or in gray? If it is green that's definitely
a bug that I'll have to look up tomorrow.

mehdideveloper, 2020-02-04:
It's in grey, but it lets me select it and play, which results in error.

BatchDrake, 2020-02-04:
Okay, I see what you mean, but I think this is bigger UX issue: here you can save multiple profiles, and since each profile has its own device, their device must show up in the device combo as well, even if it is not connected. Also, if you are plugging multiple radios in and out, this would allow switching between radios without the need of a previous refresh (which takes a lot of time). How would you do it?

mehdideveloper, 2020-02-04:
I would just disable the device when it's not connected, so the user can't open it.

BatchDrake, 2020-02-04:
However, you still need to keep the device in the list in case you save a source profile. What if we _disable_ the OK button when the settings are not valid and show a message at the bottom of the dialog explaining why? (Pretty much like in VirtualBox, see attachment) We could also complement this with a refresh button next to the device selection combo.
[image]

mehdideveloper, 2020-02-05:
In the current form, it's more like a bug. I open the dialog and it's empty. I refresh, and then a SDR which is not connected, is shown on the page and I can choose it too.

But from a user's perspective (who happens to have worked with other SDR apps), this is my personal opinion:
First, user should not be able to select the detached devices. 
Second: I think of profiles as a place to save settings (like bandwidth, sampling rate, LO freq, etc), but not necessarily the SDR name.
You might create a profile dialog like SDR Console (which may not be feasible, as you rely on SoapySDR), or you can create something like CubicSDR that only shows the connected and available devices.
In a profile, I care about all the settings, not the specific SDR (unless you have a different idea for the profiles and their use case. But that wasn't clear to me)
BTW I may attach too many SDRs when using the default profile on SigDigger. I may even attach 3 different SDRPlays. As a user, it might be confusing to me to see a dropdown with too many SDRs listed (which are not even connected) and for example 3 SDRPlays in the list which I don't know which is which, and I have to click on them to find the one that's currently connected.
What do you think? I'm not sure if I could clearly present my idea.

BatchDrake, 2020-02-05:
Yeah, I think you are right. In fact, this is somehow related to #45 , as most people assume that things should work ouf of the box. Since this is confusing and there are not too many people currently relying on profiles, I think it makes sense to refactor things a bit.

BSAsecure, 2020-04-20:
This is also related to other issues reported by the people:
https://github.com/BatchDrake/SigDigger/issues/22
https://github.com/BatchDrake/SigDigger/issues/7

Seems that automatic detection of missing SoapySDR  would improve user experience.

BatchDrake, 2020-04-20:
I see two problems here:

1. SoapySDR can only detect devices on-demand (no event system I know of)
2. SoapySDR device detection is slooow (seconds-slow).

Polling is not an option, because in some systems (e.g. my main computer) HackRF module is buggy and crashes randomly on detection. The only solution I can think of is per-OS and it would be more like a hack: I would have to use D-Bus in Linux and something similar in MacOS in order to listen to USB events, and trigger the appropriate actions. Additionally, this support should be optional as it is not required for the normal operation of SigDigger.

My concern is mostly about the reliability of this feature across systems (or even across distributions). How do we detect SDR devices? By string? ID? Every time we plug something? Also, not all devices are connected to the computer via USB (.e.g. USRP). Should we settle for a partially-working feature?

BSAsecure, 2020-04-20:
From user experience perspective, I like how SDRConsole is handling adding new devices:
[image]

You can use "best effort" to make universal detection of the devices on various OS's and Linux flavors. Detection can be triggered when user clicks "add Radio Device" which would execute "lookup" using various hacks. 

[image]

Previously defined devices can be listed and you can distinguish SDR devices based on their serial number. Not sure how USRP is detected but I guess it can be "Server" options instead of "Local" (USB, PCI based hw etc). 

For now, as SigDigger is not yet packaged into package like "apt-get install SigDigger" or single  "./install_all.sh" script. We can assume that it is power user that is capable of troubleshooting such issues. 

 May be additional refresh button that  mimics Ctrl+D behavior could improve current issue. Personally I also had to re-start SigDigger when I plug new device or in some situations when I'm changing USB port while SigDigger is running.

BatchDrake, 2020-04-20:
> You can use "best effort" to make universal detection of the devices on various OS's and Linux flavors. Detection can be triggered when user clicks "add Radio Device" which would execute "lookup" using various hacks.

I don't see exactly how this differs from the device dialog (Ctrl+D). Could you please elaborate? I think the "Refresh" button is already capable of what you have in mind.

> In SDRConsole > Select Radio, you can select device only when its detected in the system which kind of confirms that is correctly detected.

A lot of things have changed since February. Right now, only detected devices are shown in the config dialog.

> It doesn't have to be perfect: For now, as SigDigger is not yet packaged into package like "apt-get install SigDigger" or single "./install_all.sh" script. We can assume that it is power user that is capable of troubleshooting such issues.

In fact, this is not true anymore. SigDigger can be installed in Arch from the official repos using `pacman`, and it can also be built automatically using `Scripts/dist-appimage.sh`.

> May be additional refresh button that mimics Ctrl+D behavior could improve current issue. Personally I also had to re-start SigDigger when I plug new device or in some situations when I'm changing USB port while SigDigger is running.

This is definitely not normal. What SoapySDR version are you using?

BatchDrake, 2023-04-12:
I am closing this issue due to inactivity. Feel free to reopen it if this keeps not behaving as expected.

closed 2023-04-12 by BatchDrake