BatchDrake/SigDigger #57: Settings menu too small

FFY00, 2020-02-07:
Hello, the Settings menu is too narrow. It would be usable if the Y axis was resizable.

[image]

BatchDrake, 2020-02-08:
Hi,

I've just fixed this both in `master` and `develop`. Was this a custom build or did you pick an AppImage?

Cheers,

FFY00, 2020-02-08:
Hi, I built the app on Archlinux. I would like to put it in the official repos if you're okay with it.

BatchDrake, 2020-02-08:
Thank you very much. Of course I'm okay with it :) However, when you say "official repos", you mean the ones included with the distro by default (and not AUR), right? In that case, you may want to contact Viktor Drobot as he is the current maintainer of the AUR's repos (https://aur.archlinux.org/packages/sigdigger-git/)

FFY00, 2020-02-08:
Yes, exactly. My package wouldn't replace the AUR package, I would be packaging `sigdigger`, which builds from stable releases, while Viktor is packaging `sigdigger-git`, which builds from master.

BatchDrake, 2020-02-08:
In fact, `master` is now going to contain only stable releases and hotfixes, while `develop` is the testing branch with new features. SigDigger 0.1 is on the way and may be released in the next weeks, just in case you want to wait until I have something more complete. You can check https://github.com/BatchDrake/SigDigger/wiki/Roadmap-for-0.1 regularly to see how we're doing. (I will probably skip the experimental features in this release).

FFY00, 2020-02-17:
Well, I tried it yesterday and it was working properly. I will backport the patch and put 0.0.7 in the repos.

closed 2020-02-17 by FFY00