BatchDrake/SigDigger #59: Required type cast

mehdideveloper, 2020-02-10:
https://github.com/BatchDrake/SigDigger/blob/76056ef9faaa5c749cb679625968e2e95de2ae75/Components/TimeWindow.cpp#L230

`mean` and `meanSum` are `SUCOMPLEX` but `length` is defined as `int` and this causes an error:
`invalid operands to binary expression ('std::complex<float>' and 'int')`

BatchDrake, 2020-02-10:
I remember having fixing that in develop, but now develop is not building because I forgot to merge back a hotfix from master. I'll put everything in order later this evening. In the meantime, build SigDigger from `feature/ux-improvements`, SuWidgets from `develop` and suscan from `feature/ux-improvements`. That's the feature branch I'm currently working on. Sorry for breaking everything again! 😅

mehdideveloper, 2020-02-10:
Ok. I can confirm that I can compile everything in order you said and it compiles, but now I have a runtime crash!
I have created another issue for it: https://github.com/BatchDrake/SigDigger/issues/60

BatchDrake, 2020-02-10:
Just merged back to develop, everything should build now.

closed 2020-02-12 by BatchDrake