BatchDrake/SigDigger #60: Runtime crash on macOS

mehdideveloper, 2020-02-10:
This is what I get (building the latest from `feature/ux-improvements` (related to discussion from https://github.com/BatchDrake/SigDigger/issues/59)

```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [58921]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsuwidgets.1.dylib          	0x0000000106b200f8 WaveBuffer::data() const + 8
1   com.yourcompany.SigDigger     	0x000000010669645c SigDigger::TimeWindow::recalcLimits() + 44
2   com.yourcompany.SigDigger     	0x00000001066987ef SigDigger::TimeWindow::TimeWindow(QWidget*) + 175
```

BatchDrake, 2020-02-10:
Can't reproduce this one, even from `feature/ux-improvements`. How do you reproduce it? Are the branches correct? (`feature/ux-improvements` from suscan and SigDigger, develop from the rest). 

Have in mind that if you built and installed a new version of a library, you have to recompile everything that depends on that library in general.

mehdideveloper, 2020-02-10:
I cleaned everything and rebuilt all 4 repos again and still have this crash. It happens after the startup dialog is shown.
Do you think it has anything to any casts to/from SUCOMPLEX introduced in the latest changes?

BatchDrake, 2020-02-10:
Don't think so. I've checked the stacktrace and this is happening because the `WaveBuffer` object has been initialized with a wrong buffer pointer. I'm reviewing that now.

BatchDrake, 2020-02-10:
Can't figure it out. Let's try the following: go to SuWidgets, open `Waveform.cpp` and replace the method `WaveBuffer::data` with:

```const SUCOMPLEX *
WaveBuffer::data(void) const
{
  printf("This: %p\n", this);
  printf("Buffer: %p\n", this->buffer);
  return this->buffer->data();
}
```

And show me the output right before the crash.

mehdideveloper, 2020-02-10:
This is crazy!  I don't get anything on the terminal!!
It crashed with segfault but does not print out anything. Like it crashed just before entering the function body. I also tried qDebug but the same result!

BatchDrake, 2020-02-10:
This is nuts. Is it still crashing in ```WaveBuffer::data() + 8```? If that's the case, maybe it is loading the libraries from the wrong place. Try running ```otool -L SigDigger``` inside ```SigDigger.app/Contents/MacOS``` and copy the full path of ```libsuwidgets.dylib```. Then run ```strings <full path of libsuwidgets> | grep 'This:'```. You should see the strings you added.

mehdideveloper, 2020-02-10:
Yes you're right!
I just remembered why!
The reason was that SuWidgets installs the files in my Qt folder `/usr/local/Cellar/qt/5.14.0/lib` which is not in my system's path. In the past I would copy the libsuwidgets.1.dylib to /usr/local/lib to make it work (I know. Not a correct way. I think I should create a symlink instead.)
Anyway, sorry for the confusion.
BTW what do you think about the libsuwidgets install path? Why not install it in a more standard path like what you do for suscan and sigutils?

BatchDrake, 2020-02-11:
I'm glad to hear that! Regarding using standard paths, there's a complication with that. Both suscan and sigutils are built from CMake, which provides variables that expand to a nice default installation path (`/usr/local/lib` most of the time) and therefore running `make install` will work out of the box. However, SuWidgets gets built from qmake (which only provides Qt installation path variables) and the only way to make it work across systems out of the box is to install them in Qt's default installation path. Of course I could add some kind of per-system check in the .pro files (in fact that's how it used to be in the past), but that's not maintainable and will break sooner or later when we face a guy with a completely absurd system configuration.

A workaround for this is to pass the installation path of SuWidgets in the command line when you run qmake (that's what I use for deployment), but maybe it's too cumbersome and you may want to write an alias or something.

closed 2020-02-12 by BatchDrake