BatchDrake/SigDigger #61: Settings dialog resize

mehdideveloper, 2020-02-11:
Do you need to allow the user to resize the settings dialog? It's not a big deal, but usually these dialogs are fixed size, as there's no need to change the size. I'm not sure if it was like this from the first day, or you changed it because of #57

BatchDrake, 2020-02-11:
Well, it shouldn't and if it is the case, it is certainly a UI bug.

mehdideveloper, 2020-02-11:
Look at the two screenshots (original, and resized)
[image]
[image]

BatchDrake, 2020-02-12:
It should have been fixed now in feature/ux-improvements

mehdideveloper, 2020-02-13:
It is fixed 👍

closed 2020-02-13 by BatchDrake