BatchDrake/SigDigger #66: New crash on macOS (after switching SDRs)

mehdideveloper, 2020-02-15:
Hi
This is a different crash than #64 
How to reproduce:
1- I connected a LimeSDR Mini and tried to run it. I got an error, because a bug in SoapySDR prevents me from selecting the RX antenna on the LimeSDR Mini and I see an error dialog.
2- Now I unplug the LimeSDR Mini, connect an AirSpy and go to View->Devices and refresh so the newly connected SDR appears in the list.
3- Now I go to Settings and select AirSpy
4- Click on start

I see a crash dialog and what's strange is that it says the connection to LimeSDR Mini can't be started! It's like trying to resume the previous session!
The error is `Failed to open SDR device: Failed to make a connection with 'LimeSDR Mini'`

BatchDrake, 2020-02-15:
Is this a crash or a startup error? Does the program close itself after that message?

mehdideveloper, 2020-02-16:
Sorry for the bad wording. Just edited the title. It is not a crash. It's just an error dialog

BatchDrake, 2020-02-16:
Christ, I hate LimeSDR Mini so much. After a bunch of crashes and wasted time I got to make it work (choppy audio, can't get more than 11 Msps with it) and reproduced your error. Looking into it now.

BatchDrake, 2020-02-16:
Done. You can try again now.

closed 2023-04-12 by BatchDrake