BatchDrake/SigDigger #67: Compile error because of FrequencyAllocationTable

mehdideveloper, 2020-02-16:
Hey
So I tried to build the latest (from ux-improvements) on 2 machines. I also rebuilt all the other 3 repos.
On machine 1, it works. On the second machine, the other 3 repos build successfully but SigDigger gives me this compile error:

```
include/MainSpectrum.h:54:17: error: use of undeclared identifier 'FrequencyAllocationTable'
    std::vector<FrequencyAllocationTable *> FATs;
                ^
include/MainSpectrum.h:54:43: error: expected expression
    std::vector<FrequencyAllocationTable *> FATs;
                                          ^
include/MainSpectrum.h:72:12: error: unknown type name 'FrequencyBand'
    static FrequencyBand deserializeFrequencyBand(Suscan::Object const &);
           ^
include/MainSpectrum.h:103:18: error: unknown type name 'FrequencyAllocationTable'
    void pushFAT(FrequencyAllocationTable *);
                 ^
include/MainSpectrum.h:116:5: error: unknown type name 'FrequencyAllocationTable'
    FrequencyAllocationTable *getFAT(QString const &) const;
```

Any ideas?

BatchDrake, 2020-02-16:
Yes. Either missing make install, wrong installation directories or improper branch. Go to SuWidgets and run:

```grep FrequencyAllocationTable *.cpp```

If this does not show any message, you have an out-of-date directory, or you are in the wrong branch. If you do, look for SigDigger's include paths during build, and grep the same thing in Waterfall.h. If you don't get anything, you either did not run make install or the install path is wrong (which could happen if you have compiled both SuWidgets and SigDigger with different Qt installations). If you still see these errors, you have an old SuWidgets directory installed somewhere in your system's include directories that is shading the current version.

mehdideveloper, 2020-02-16:
You're right. My mistake :(
But now I have another issue: linker error in SigDigger:

```
ld: dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib missing LC_ID_DYLIB load command file '/usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

I have built SuWidgets from ux-improvements and installed, before making SigDigger

mehdideveloper, 2020-02-16:
Hmmmmm. Interesting:
```
/Library/Developer/CommandLineTools/usr/bin/objdump: error: '/usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib': truncated or malformed object (load command 7 LC_LOAD_DYLIB library name extends past the end of the load command)
```

BatchDrake, 2020-02-16:
This means that the build was aborted in the middle of the linking phase. The library file was produced nonetheless, so make install assumed it was properly built and installed nonetheless.

mehdideveloper, 2020-02-16:
Ok this is weird!
When I install SuWidgets with `make install` (which actually creates symlinks to libraries) this issue happens, but when I copy the .dylib files manually to the Qt lib folder, everything is fine!!
Anyway, with this trick I managed to build the SigDigger successfully on this machine

BatchDrake, 2020-02-20:
Is this still happening?

mehdideveloper, 2020-02-22:
Yes!
I have no idea why.

BatchDrake, 2020-02-22:
Could you attach the output of SuWidgets' make install and the command you use to copy the library?

mehdideveloper, 2020-02-22:
This is the output of `make install` for SuWidgets:
```
( test -e Makefile.SuWidgetsLib || /usr/local/Cellar/qt/5.14.0/bin/qmake -o Makefile.SuWidgetsLib /Users/user1/Documents/Source/SuWidgets/SuWidgetsLib.pro ) && /Library/Developer/CommandLineTools/usr/bin/make -f Makefile.SuWidgetsLib install
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.SuWidgetsLib.Release install
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall -exe libsuwidgets.1.0.0.dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.1.0.0.dylib
strip -S -x /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.1.0.0.dylib
ln -f -s libsuwidgets.1.0.0.dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib
ln -f -s libsuwidgets.1.0.0.dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.1.dylib
ln -f -s libsuwidgets.1.0.0.dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.1.0.dylib
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/ctkPimpl.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/ctkPimpl.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/ctkRangeSlider.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/ctkRangeSlider.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/LCD.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/LCD.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Histogram.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Histogram.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Constellation.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Constellation.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/QVerticalLabel.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/QVerticalLabel.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/SymView.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/SymView.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Decider.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Decider.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Transition.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Transition.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Waveform.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Waveform.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/Waterfall.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/Waterfall.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/ColorChooserButton.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/ColorChooserButton.h
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall /Users/user1/Documents/Source/SuWidgets/ThrottleableWidget.h /usr/local/Cellar/qt/5.14.0/include/SuWidgets/ThrottleableWidget.h
( test -e Makefile.SuWidgetsPlugin || /usr/local/Cellar/qt/5.14.0/bin/qmake -o Makefile.SuWidgetsPlugin /Users/user1/Documents/Source/SuWidgets/SuWidgetsPlugin.pro ) && /Library/Developer/CommandLineTools/usr/bin/make -f Makefile.SuWidgetsPlugin install
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.SuWidgetsPlugin.Release install
/usr/local/Cellar/qt/5.14.0/bin/qmake -install qinstall -exe libsuwidgetsplugin.dylib /usr/local/Cellar/qt/5.14.0/plugins/designer/libsuwidgetsplugin.dylib
strip -S -x /usr/local/Cellar/qt/5.14.0/plugins/designer/libsuwidgetsplugin.dylib
```

After this I build SigDigger but it fails with:
```
ld: dylib /usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib missing LC_ID_DYLIB load command file '/usr/local/Cellar/qt/5.14.0/lib/libsuwidgets.dylib' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Then I do this in the SuWidget folder:
```
cp libsuwidgets* /usr/local/Cellar/qt/5.14.0/lib/
```

Then SigDigger make will finish successfully.

BatchDrake, 2020-02-22:
I think I've found the reason: https://github.com/Homebrew/homebrew-core/issues/47315

mehdideveloper, 2020-02-22:
You're right! It solved my issue. 
Thanks

closed 2020-02-23 by BatchDrake