BatchDrake/SigDigger #69: Selected signal frequency

mehdideveloper, 2020-02-18:
Hi
I have a suggestion: when the user selects a signal in the waterfall, its frequency should be shown somewhere. Otherwise it would be impossible to know the exact frequency. The frequency label in the top-left just shows the centre frequency.

BatchDrake, 2020-02-19:
It is already shown in the Inspector tab, although maybe it would be a better idea to show it somewhere more accessible, as the status window.

BatchDrake, 2020-02-19:
I wrote early this morning some tentative code to handle that. I think it may make sense to put some extra information in the status bar like sample rate, demod or things of the sort. What do you think?

BatchDrake, 2020-02-19:
(Forgot to mention, this is not pushed yet, I'll commit the changes later today)

mehdideveloper, 2020-02-19:
> I wrote early this morning some tentative code to handle that. I think it may make sense to put some extra information in the status bar like sample rate, demod or things of the sort. What do you think?

The demodulation, sample rate, etc are already shown in another pane/window, so I'm not sure if it adds any value to the user experience, but the selected frequency is very important (all the SDR apps that I know, either only show the selected frequency, or show the selected and centre frequency separately)

BatchDrake, 2020-02-19:
Got it. I'm currently placing it in the bottom right corner of the window, what would be for you the best place to put it?

mehdideveloper, 2020-02-19:
Well I'm used it to be in the top of the screen (usually centre or left)
I don't know any SDR app that puts it in the bottom, so I think it would be good to follow other apps and make it easy for new users

BatchDrake, 2020-02-19:
How about replacing the LO LCD (right to the center frequency) by the full selected frequency? The information it provides is basically the same.

mehdideveloper, 2020-02-19:
I think this is the best option 👍

BatchDrake, 2020-02-20:
Done, pull from the usual branches.

mehdideveloper, 2020-02-21:
Thanks. I think it would be better if it has a bigger font. 
What's the purpose of always showing the centre frequency? I mean, it can be either slower (because it's less important), or it can be removed altogether.
When I select on a specific signal, I just look for the frequency. What do you think?

BatchDrake, 2020-02-21:
I don't agree. When I use the device, I want to know exactly at what frequency it has been tuned. If I change that number, I want to know that that is what I am requesting to the device. This is also the number used to name capture files. This is by design, as I found the behavior you describe in Gqrx rather confusing.

BatchDrake, 2020-02-21:
Maybe the best approach here could be add a GUI option to control the behavior of the LCD, but I think it's better to leave that for a next release now.

mehdideveloper, 2020-02-21:
I think we are saying the same thing!
I double checked Gqrx, SDR# and SDR Console. They all just show the tuned frequency. Right now SigDigger shows the centre frequency (which is fixed) in a big font, and the tuned frequency in a smaller font. 
My suggestion was that to make the tuned frequency LCD font bigger (because it's more important)

mehdideveloper, 2020-02-21:
[image]
 This is small in my opinion

BatchDrake, 2020-02-21:
I agree that for some scenarios having the main LCD show the demod frequency makes sense. Still, there are several reasons why I'd like to keep the current display:

1. One of the features that have been present in the old Suscan's GUI and have not been ported (yet) to SigDigger is the channel detector (which, interestingly, was one of the reasons why I coded Suscan to begin with). The most relevant use case was "detect all channels around this frequency" (the one displayed in the LCD) or "fingerprint this frequency". In the end, Suscan works with a portion of the spectrum tagged by the device frequency, and that is what is shown in the LCD.

2. It shows the frequency that has been passed to the device and what part of the spectrum I am seeing. Gqrx's "tuned frequency" is actually the demodulator frequency, which is a number constructed by adding the relative channel frequency (also called LO frequency, in the range [-sampRate/2, sampRate/2) to the device frequency, and only exists inside my computer. 

3. It is not ambiguous. In Gqrx, since the same so-called "tuner" frequency may refer to different hardware frequencies, setting that number to a frequency you recall may lead to a completely different RF setting and produce confusing results. For instance: HackRF's image rejection is rather poor, and if you don't adjust your filters right, you will see shifted reversed copies of nearby powerful signals everywhere. The same Gqrx tuned frequency may show these images in different places which in my opinion is counter-intuitive. This could be mitigated to some extent by changing the device frequency in an informed way (which is what SigDigger allows you to do unambiguously): if you tune your device to frequency F, you will have images but if you tune it to frequency F+delta, you won't (and this is the setting that gets saved).

4. Gqrx's LCD does two different things at a time: it lets you _indirectly_ change the device frequency while showing you the quantity hardware frequency + channel frequency. I.e. it displays the sum of two values, while its modification only changes one of them. On the other hand, SigDigger's LCD allows you to do only one thing: adjust the device frequency.

I think the best we can do about it is to keep this as an improvement for the next release, configurable through the settings window. What do you think?

closed 2023-04-12 by BatchDrake