BatchDrake/SigDigger #70: Check the minimum and maximum frequency

mehdideveloper, 2020-02-21:
I have a suggestion:
check the upper and lower supported frequencies for the connected device and don't allow the user to tune to anything outside the range (neither show them in the spectrum)
I think this is feasible using SoapySDR's API.
I tried to test the latest commits, and the frequency was 0 (on the settings pane). I'm not sure if it was always like this, or it's new. Anyway, I clicked on the start and then the spectrum was showing the centre frequency of 0 and negative frequencies to the left ! (The connected device was AirSpy one which does not support below 24MHZ)

BatchDrake, 2020-02-21:
That's indeed a bug. The frequency is not being refreshed in the settings dialog. Also, the arrow symbols don't show in all machines. I will have to use symbolic icons here.

mehdideveloper, 2020-02-25:
This would be my top request (as a regular user)  to be fixed.
It's annoying to set the frequency everytime I fire up the app to do some tests

BatchDrake, 2020-02-25:
I am failing to reproduce this at home, still looking into it.

mehdideveloper, 2020-02-25:
Screenshot for reference:

[image]

BatchDrake, 2020-02-26:
I think I fixed it. Pull the latest changes from ux-improvements as usual.

mehdideveloper, 2020-02-26:
Yes it is fixed 👍

closed 2020-02-26 by BatchDrake