BatchDrake/SigDigger #71: Two options for full screen on macOS

mehdideveloper, 2020-02-21:
Hey
On macOS there are 2 options in the View menu to go full screen. Look at the attached screenshot:

[image]

BatchDrake, 2020-02-21:
Dang. I'd like to keep the F11 behavior as it would work in a platform-independent way. Some people found a solution using Objective C, but maybe is too cumbersome: https://forum.qt.io/topic/60623/qt-5-4-2-os-x-10-11-el-capitan-how-to-remove-the-enter-full-screen-menu-item/6

BatchDrake, 2020-02-22:
Just took the path of least resistance and removed the full screen option completely in MacOS X. Changes in the usual branches.

mehdideveloper, 2020-02-22:
Fixed 👍

closed 2020-02-22 by BatchDrake