BatchDrake/SigDigger #72: Adjusting the visual gain

mehdideveloper, 2020-02-22:
When working on weak signals, we need to have an option to either increase the visual gain or expand the spectrum to -100dB (instead of -50)
For example look at the attached screenshot: I am trying to receive some weak signals with AirSpy HF+ and even with the maximum gain settings, signals can't be properly shown.
 
[image]

BatchDrake, 2020-02-22:
You already can. Use the mouse wheel on the left side of the spectrum.
There should also be a control in the FFT panel calles "panadapter range".

mehdideveloper, 2020-02-22:
Oh! I never used it 🙈 
Thanks

closed 2020-02-23 by BatchDrake