BatchDrake/SigDigger #73: Filter settings

mehdideveloper, 2020-02-23:
Is there a way to adjust filter bandwidth in SigDigger?
Look at the attached screenshot. The WFM signal had a bigger bandwidth of what's highlighted in the spectrum. It should be very nice if user can change the filter bandwidth. Or maybe some default pre-sets would be good as well. Like 2.7KHZ for SSB, something between 6-9KHZ for AM and 200KHZ(?!) for FM. 
Which also brings me to the 2nd question: have you considered having two separate NFM and WFM  demodulators?

mehdideveloper, 2020-02-23:
[image]

mehdideveloper, 2020-02-23:
This is crazy! I have been using this app for a while and hadn't noticed that you can change the width with mouse !!

BatchDrake, 2020-02-23:
Dude 😂😂😂

BatchDrake, 2020-02-23:
I definitely have to prepare a small manual with the next release.

mehdideveloper, 2020-02-23:
I will help you with the manual

BatchDrake, 2020-02-23:
I've just opened an issue for that: https://github.com/BatchDrake/SigDigger/issues/74

closed 2020-02-23 by BatchDrake