BatchDrake/SigDigger #75: White noise after enabling audio

mehdideveloper, 2020-02-24:
I am not sure if this is a bug in the AudioPlayback or somewhere else in SigDigger, but report it here for the record:
1- Tune to a station frequency, while the audio demod is disabled
2- Enable the audio demod. You will only hear white noise.
3- Change the frequency a little bit (like click on a section in spectrum, very close to the tuned frequency) and you will hear the correct demodulated audio.

This also happens when you change the demodulator while audio demod is enabled. (This does not happen when changing audio sample rates on the fly)

BatchDrake, 2020-02-24:
This probably has to do with an improper LO setting on start. Hopefully has an easy fix.

BatchDrake, 2020-02-24:
Turns it out was related to a signedness issue while checking audio inspector limits. Try again now.

mehdideveloper, 2020-02-25:
Thanks. Fixed 👍

closed 2020-02-26 by BatchDrake