BatchDrake/SigDigger #86: Bug in recording

mehdideveloper, 2020-03-05:
When recording an audio or I/Q recording, if we select a folder which we don't have permission to write to, the app's behaviour is non-consistent:

1- For recording I/Q, the label of the button is changed from Record to Stop, but an error dialog is shown : 
> Failed to open capture file for writing: Read-only file system
I think changing the label should be done after the file is opened successfully.

2- For recording audio, no error is shown and the label of the button is changed from Record to Stop, but still the size of the recording is 0. I think the same fix as above should be applied here as well.

BatchDrake, 2020-03-06:
This should be fixed now.

mehdideveloper, 2020-03-09:
fixed 👍

closed 2020-03-09 by BatchDrake