BatchDrake/SigDigger #90: error in make in SigDigger

fe8769a, 2020-03-18:
Hello

Doing a make while compiling Sigdigger , I have the following error :./ui_Config.h: In member function ‘void Ui_Config::retranslateUi(QDialog*)’:
./ui_Config.h:822:23: error: ‘class QComboBox’ has no member named ‘setCurrentText’; did you mean ‘setCurrentIndex’?
         profileCombo->setCurrentText(QString());
                       ^~~~~~~~~~~~~~
                       setCurrentIndex
Makefile:648: recipe for target 'AppConfig.o' failed
make: *** [AppConfig.o] Error 1
I didn't find any explaination 

OS is Xubuntu 18.04.4 

Thanks for help

BatchDrake, 2020-03-19:
Hi, thank you for your feedback. What Qt version are you using?

fe8769a, 2020-03-24:
Please close this call . I did a new reinstall

closed 2020-03-24 by BatchDrake