BatchDrake/SigDigger #91: compile error in macOS

mehdideveloper, 2020-03-19:
There's a new mismatch between two operands in the recent changes in develop:
`CarrierDetector.cpp` line 133:
`acc += psd * SU_C_EXP(I * M_PI * nFreq);`
`I` is std::complex. Changing it to something like 
`acc += psd * SU_C_EXP(SU_C_REAL(I) * M_PI * nFreq);`
fixes

BatchDrake, 2020-03-19:
The problem is that that change, mathematically, does not make sense. SU_C_EXP must accept the imaginary unit times pi times `nFreq`.

BatchDrake, 2020-03-20:
Also, I need the specific error.

mehdideveloper, 2020-03-27:
Sorry for the late reply. I think you solved this, right? I don't get errors anymore (pulled the latest changes)

BatchDrake, 2020-03-27:
Yep, days ago. Anyways, get ready because there's a big feature in the oven and a lot of things may break in the following days.

closed 2020-03-27 by BatchDrake