BatchDrake/SigDigger #92: error in Make in sigDigger

fe8769a, 2020-03-24:
After a complete xubuntu 18.4.4 reinstall and gnuradio and hardware , upgrade to qt 5.12 and qmake associated . I have this error nearly at the end of make in Sigdigger :

g++ -Wl,-O1 -o SigDigger AppConfig.o Application.o AppUI.o ColorConfig.o Loader.o AudioPlayback.o AboutDialog.o AudioPanel.o ConfigDialog.o DataSaverUI.o DeviceGain.o FftPanel.o GainSlider.o InspectorPanel.o MainSpectrum.o MainWindow.o PersistentWidget.o SaveProfileDialog.o SourcePanel.o Inspector.o InspectorUI.o AfcControl.o AskControl.o ClockRecovery.o EqualizerControl.o GainControl.o InspectorCtl.o MfControl.o ToneControl.o AutoGain.o Averager.o Palette.o SNREstimator.o ChannelMessage.o GenericMessage.o InspectorMessage.o PSDMessage.o SamplesMessage.o Analyzer.o AnalyzerParams.o Config.o Exception.o Library.o Logger.o Message.o MQ.o Object.o Serializable.o Source.o AudioMediator.o FftMediator.o InspectorMediator.o SourceMediator.o SpectrumMediator.o UIMediator.o main.o EstimatorControl.o GenericDataSaver.o FileDataSaver.o SocketForwarder.o NetForwarderUI.o WaitingSpinnerWidget.o DeviceDialog.o DeviceDialogMediator.o qrc_Icons.o moc_Analyzer.o moc_AboutDialog.o moc_AfcControl.o moc_Application.o moc_AskControl.o moc_AudioPanel.o moc_AudioPlayback.o moc_ClockRecovery.o moc_ConfigDialog.o moc_DataSaverUI.o moc_DeviceGain.o moc_EqualizerControl.o moc_FftPanel.o moc_GainControl.o moc_GainSlider.o moc_Inspector.o moc_InspectorCtl.o moc_InspectorPanel.o moc_InspectorUI.o moc_Loader.o moc_MainSpectrum.o moc_MainWindow.o moc_MfControl.o moc_PersistentWidget.o moc_SaveProfileDialog.o moc_SourcePanel.o moc_ToneControl.o moc_UIMediator.o moc_EstimatorControl.o moc_GenericDataSaver.o moc_FileDataSaver.o moc_SocketForwarder.o moc_NetForwarderUI.o moc_WaitingSpinnerWidget.o moc_DeviceDialog.o   -L/usr/lib/x86_64-linux-gnu -lsuwidgets -L/usr/local/lib -lsuscan -lsigutils -lm -lfftw3f -lvolk -lfftw3 -lsndfile -lSoapySDR -lxml2 -lasound -lQt5Widgets -lQt5Gui -lQt5Core -lGL -lpthread 
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QColorDialog::getColor(QColor const&, QWidget*) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QMapData::createData(int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QByteArray::realloc(int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QVectorData::shared_null »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QMapData::continueFreeData(int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QVectorData::grow(int, int, int, bool) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QMapData::shared_null »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QVectorData::free(QVectorData*, int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QChar::QChar(char) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::toAscii() const »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::free(QString::Data*) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QCoreApplication::translate(char const*, char const*, char const*, QCoreApplication::Encoding) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QVectorData::allocate(int, int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QVectorData::reallocate(QVectorData*, int, int, int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QMapData::node_create(QMapData::Node**, int, int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::arg(double, int, char, int, QChar const&) const »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::arg(long long, int, int, QChar const&) const »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « qt_message_output(QtMsgType, char const*) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QMetaObject::tr(char const*, char const*) const »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::fromUtf8(char const*, int) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QPainter::initFrom(QWidget const*) »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::shared_null »
/usr/lib/x86_64-linux-gnu/libsuwidgets.so : référence indéfinie vers « QString::arg(QString const&, int, QChar const&) const »
collect2: error: ld returned 1 exit status
Makefile:475: recipe for target 'SigDigger' failed
make: *** [SigDigger] Error 1

any idea ?

Thanks

BatchDrake, 2020-03-24:
It can't be the Qt version then, I have Qt 5.12.5 in the computer I use for development. 

Could it be that you built SuWidgets with an older version of Qt, installed it through `sudo make install`, upgraded to 5.12 after the errors you got in your previous issue and then tried to rebuild again? If that's the case, my advice would be to perform a complete build/install of both SuWidgets and SigDigger by cloning them again in a clean directory and see whether this keeps happening.

BatchDrake, 2020-03-24:
Yep, it totally looks like that, because my `libsuwidgets.so` has no calls to the methods g++ is complaining about whatsoever.

fe8769a, 2020-03-25:
rebuild from scratch SuWidgets and SigDigger , ,working fine 

Thanks for help

closed 2020-03-27 by BatchDrake