BatchDrake/SigDigger #93: Sigdigger not start

fe8769a, 2020-03-25:
when starting sigdigger , the following error is displayed :+1: 

Failed to start capture due to errors:
error: Failed to open SDR device: RuntimeError: Expected FPGA compatibility number 14, but got 16:
The FPGA build is not compatible with the host code build.
Please run:

 "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"
error: exception in "suscan_source_open_sdr(new)" (analyzer/source.c:2081)
error: exception in "analyzer->source = suscan_source_new(config)" (analyzer/analyzer.c:737)
error: Failed to initialize source

uhd_images_downloader.py is in the right directory

result of suscan status

suscan.status 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[INFO] Detected Device: B210
[INFO] Operating over USB 3.
suscan.status: suscan library loaded successfully.
---------8<-------------------------------------
(e) exception in "sdev = SoapySDRDevice_make(dev->args)" (analyzer/source.c:222)
(e) exception in "suscan_source_device_populate_info(dev)" (analyzer/source.c:552)

BatchDrake, 2020-03-27:
This seems a UHD problem, not SigDigger's. Look at this message:

```
error: Failed to open SDR device: RuntimeError: Expected FPGA compatibility number 14, but got 16:
The FPGA build is not compatible with the host code build.
```

Please try executing:

```
/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py
```

And tell what the output is (I don't know whether you may need to run this as root).

BatchDrake, 2023-04-12:
I am closing this issue due to inactivity. Feel free to reopen it if the problem persists.

Cheers,

closed 2023-04-12 by BatchDrake