BatchDrake/SigDigger #95: core dump with SigDigger 0.1

fe8769a, 2020-04-11:
Hello 

tryed new version with B210 ettus . Previous version was working fine 

Stop with core dump error

/opt/SigDigger/bin/SigDigger
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.002-0-gf18abe54

[INFO] Detected Device: B210
[INFO] Operating over USB 3.
[INFO] Initialize CODEC control...
[INFO] Initialize Radio control...
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
Error: C++ API exception: expression `obj = suscan_object_get_field(this->instance, field.c_str())' failed in include/Suscan/Object.h:173
[INFO] Detected Device: B210
[INFO] Operating over USB 3.
[INFO] Initialize CODEC control...
[INFO] Initialize Radio control...
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
[INFO] Asking for clock rate 32.000000 MHz... 
[INFO] 
Actually got clock rate 32.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
Erreur de segmentation (core dumped)

BatchDrake, 2020-04-11:
Hi,

Is this one of the precompiled AppImages in the latest release? If that's the case, are you running the `lite` or the `full` version? Which version was the previous one you were using?

If you've built SigDigger from sources, please double check that you've built `sigutils`, `suscan`, `SuWidgets` and `SigDigger` from the `master` branch, in this precise order. Have in mind that you have to run `make install` in every subproject _before_ moving to the next one. 

If this error persists, could you send me a backtrace of the crash? You can do it with `gdb` by performing the following steps (you may need to install it from your GNU/Linux distribution repositories):

1. Open a terminal and go to SigDigger's directory (basically where the SigDigger executable is)
2. Run `gdb ./SigDigger`, you should see a lot of copyright strings and a prompt saying `(gdb)`. 
3. In the `(gdb)` prompt, type `run`. SigDigger will atempt to start.
4. After reproducing the crash, you should see the `(gdb)` prompt again in your terminal window. Then, type `bt`
5. A list of function calls should show up. Copy and paste it right here.

Thanks!

fe8769a, 2020-04-12:
Hello
Previous version was working fine - compiled from source and lite or full version

Now 

With the full version I have :+1: 

./SigDigger-0.1.0-x86_64-full.AppImage 
QApplication: invalid style override passed, ignoring it.
/tmp/.mount_SigDigcXv1nG/usr/bin/SigDigger.app: symbol lookup error: /tmp/.mount_SigDigcXv1nG/usr/bin/../lib/libQt5Gui.so.5: undefined symbol: hb_font_funcs_set_font_h_extents_func

With lite version :+1: 

The same message

With the version builded from source the error core dump 

I did as same I did for the working version 

Make , make install et sudo ldconfig each time 

here is the gdb result

gdb /opt/SigDigger/bin/SigDigger
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/SigDigger/bin/SigDigger...(no debugging symbols found)...done.
(gdb) run
Starting program: /opt/SigDigger/bin/SigDigger 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe9b4d700 (LWP 3070)]
[New Thread 0x7fffdd2b1700 (LWP 3073)]
[New Thread 0x7fffd6bdf700 (LWP 3074)]
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.002-0-gf18abe54

[New Thread 0x7fffccb16700 (LWP 3075)]
[Thread 0x7fffccb16700 (LWP 3075) exited]
[New Thread 0x7fffccb16700 (LWP 3076)]
[Thread 0x7fffccb16700 (LWP 3076) exited]
[New Thread 0x7fffccb16700 (LWP 3077)]
[New Thread 0x7fffcc315700 (LWP 3078)]
[New Thread 0x7fffcbb14700 (LWP 3079)]
[New Thread 0x7fffcb313700 (LWP 3080)]
[Thread 0x7fffcb313700 (LWP 3080) exited]
[New Thread 0x7fffcb313700 (LWP 3081)]
[Thread 0x7fffcb313700 (LWP 3081) exited]
[New Thread 0x7fffcb313700 (LWP 3082)]
[Thread 0x7fffcb313700 (LWP 3082) exited]
[New Thread 0x7fffcb313700 (LWP 3083)]
-- Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
[Thread 0x7fffcb313700 (LWP 3083) exited]
[New Thread 0x7fffcb313700 (LWP 3123)]
[Thread 0x7fffcb313700 (LWP 3123) exited]
[New Thread 0x7fffcb313700 (LWP 3124)]
[Thread 0x7fffcb313700 (LWP 3124) exited]
[New Thread 0x7fffcb313700 (LWP 3125)]
[Thread 0x7fffcb313700 (LWP 3125) exited]
[... 152 lines cut ...]
[INFO] Performing register loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Performing CODEC loopback test... 
[INFO] pass
[INFO] Asking for clock rate 16.000000 MHz... 
[INFO] 
Actually got clock rate 16.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Setting master clock rate selection to 'automatic'.
[INFO] Asking for clock rate 32.000000 MHz... 
[INFO] 
Actually got clock rate 32.000000 MHz.
[INFO] Performing timer loopback test... 
[INFO] pass
[INFO] Performing timer loopback test... 
[INFO] pass

Thread 1 "SigDigger" received signal SIGSEGV, Segmentation fault.
0x00007ffff6e87c16 in SoapySDRDevice_setupStream ()
   from /usr/local/lib/libSoapySDR.so.0.7

BatchDrake, 2020-04-12:
Okay, now in gdb, after the SIGSEGV you have to type `bt` and press enter. That is the output that I need.

Also, what GNU/Linux distribution are you using?

Thanks,

fe8769a, 2020-04-12:
I'm using ubuntu 16.4.1 and gnu radio 3.7.9 

BT results

#0  0x00007ffff6e87c16 in SoapySDRDevice_setupStream ()
   from /usr/local/lib/libSoapySDR.so.0.7
#1  0x00007ffff7964e3e in suscan_source_open_sdr (source=0x102e8c0)
    at /home/fe8769/Sigdigger/suscan/analyzer/source.c:1856
#2  0x00007ffff7965e29 in suscan_source_new (config=0x12751d0)
    at /home/fe8769/Sigdigger/suscan/analyzer/source.c:2308
#3  0x00007ffff7958129 in suscan_analyzer_source_init (analyzer=0x15e8490, 
    config=0x12751d0) at /home/fe8769/Sigdigger/suscan/analyzer/analyzer.c:854
#4  0x00007ffff7958380 in suscan_analyzer_new (params=0x7fffffffca38, 
    config=0x12751d0, mq=0x125a270)
    at /home/fe8769/Sigdigger/suscan/analyzer/analyzer.c:933
#5  0x00000000004a8d67 in ?? ()
#6  0x000000000042cc25 in ?? ()
#7  0x00000000004d8a52 in ?? ()
#8  0x00007ffff5d44c0a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00000000004dd918 in ?? ()
#10 0x00007ffff5d44c0a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff65ff412 in QAction::triggered(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff6601898 in QAction::activate(QAction::ActionEvent) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
---Type <return> to continue, or q <return> to quit---
#13 0x00007ffff67075a0 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff67076d4 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff67cc26a in QToolButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff664bfc8 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff67cc349 in QToolButton::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff660905c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff660ec19 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff5d1626b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff660db32 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff66665bb in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff6668b7b in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff660905c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff660e516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff5d1626b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff60584e1 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007ffff605a1a5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#29 0x00007ffff603df08 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#30 0x00007ffff7f0b200 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#31 0x00007ffff33dd197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff33dd3f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff33dd49c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff5d6c6af in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff5d13a2a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff5d1bacc in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x000000000042447c in ?? ()
#38 0x00007ffff4c28830 in __libc_start_main (main=0x424410, argc=1, argv=0x7fffffffdf18, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdf08) at ../csu/libc-start.c:291
#39 0x0000000000424579 in ?? ()

BatchDrake, 2020-04-13:
This is odd. The crash is somehow happening inside the SoapySDR wrapper, but I haven't changed anything there.

There should be a config file in your home folder named `/home/fe8769/.suscan/config/uiconfig.xml`. Could you attach it to this issue? Also, what SDR are you using?

fe8769a, 2020-04-14:
I'm using Ettus B210

here is uiconfig.xlm content

<?xml version="1.0" ?>

<suscan:serialization xmlns:suscan="http://actinid.org/suscan" name="root">
  <suscan:object class="qtui">
    <suscan:field name="width" value="1920" />
    <suscan:field name="height" value="1031" />
    <suscan:field name="x" value="0" />
    <suscan:field name="y" value="49" />
    <suscan:field name="loFreq" value="58084" />
    <suscan:field name="bandwidth" value="32932" />
    <suscan:object name="source" class="source_config">
      <suscan:field name="type" value="SDR" />
      <suscan:field name="antenna" value="RX2" />
      <suscan:field name="freq" value="1084820992.000000" />
      <suscan:field name="lnb_freq" value="0.000000" />
      <suscan:field name="bandwidth" value="200000.000000" />
      <suscan:field name="iq_balance" value="false" />
      <suscan:field name="dc_remove" value="true" />
      <suscan:field name="loop" value="true" />
      <suscan:field name="samp_rate" value="1000000" />
      <suscan:field name="average" value="1" />
      <suscan:field name="channel" value="0" />
      <suscan:object name="sdr_args">
        <suscan:field name="driver" value="uhd" />
        <suscan:field name="label" value="B210 86680AA" />
        <suscan:field name="name" value="Zhixun-wireless_B210" />
        <suscan:field name="product" value="B210" />
        <suscan:field name="serial" value="86680AA" />
        <suscan:field name="type" value="b200" />
      </suscan:object>
      <suscan:object name="gains">
        <suscan:field name="PGA" value="63.000000" />
      </suscan:object>
    </suscan:object>
    <suscan:object name="analyzerParams" class="AnalyzerParams">
      <suscan:field name="channelUpdateInterval" value="0.100000" />
      <suscan:field name="psdUpdateInterval" value="0.040000" />
      <suscan:field name="spectrumAvgAlpha" value="0.010000" />
      <suscan:field name="sAvgAlpha" value="0.001000" />
      <suscan:field name="nAvgAlpha" value="0.500000" />
      <suscan:field name="snr" value="2.000000" />
      <suscan:field name="windowSize" value="8192" />
      <suscan:field name="windowFunction" value="blackmann-harris" />
    </suscan:object>
    <suscan:object name="colors" class="colorcfg">
      <suscan:field name="lcdBackground" value="#1d1d1f" />
      <suscan:field name="lcdForeground" value="#b7e869" />
      <suscan:field name="spectrumBackground" value="#1d1d1f" />
      <suscan:field name="spectrumForeground" value="#ffffff" />
      <suscan:field name="spectrumAxes" value="#808080" />
      <suscan:field name="spectrumText" value="#ffffff" />
      <suscan:field name="constellationForeground" value="#b7e869" />
      <suscan:field name="constellationBackground" value="#1d1d1f" />
      <suscan:field name="constellationAxes" value="#808080" />
      <suscan:field name="transitionForeground" value="#b7e869" />
      <suscan:field name="transitionBackground" value="#1d1d1f" />
      <suscan:field name="transitionAxes" value="#808080" />
      <suscan:field name="histogramForeground" value="#b7e869" />
      <suscan:field name="histogramBackground" value="#1d1d1f" />
      <suscan:field name="histogramAxes" value="#808080" />
      <suscan:field name="histogramModel" value="#ffff00" />
    </suscan:object>
    <suscan:object name="sourcePanel" class="SourcePanelConfig">
      <suscan:field name="throttle" value="false" />
      <suscan:field name="throttleRate" value="196000" />
      <suscan:field name="captureFolder" value="/media/fe8769/test_vm/21032020" />
      <suscan:field name="dcRemove" value="true" />
      <suscan:field name="iqRev" value="false" />
      <suscan:field name="agcEnabled" value="false" />
    </suscan:object>
    <suscan:object name="fftPanel" class="FftPanelConfig">
      <suscan:field name="averaging" value="1.000000" />
      <suscan:field name="panWfRatio" value="0.300000" />
      <suscan:field name="peakDetect" value="false" />
      <suscan:field name="peakHold" value="false" />
      <suscan:field name="panRangeMin" value="-44.000000" />
      <suscan:field name="panRangeMax" value="-8.000000" />
      <suscan:field name="wfRangeMin" value="-60.000000" />
      <suscan:field name="wfRangeMax" value="-10.000000" />
      <suscan:field name="palette" value="Suscan" />
      <suscan:field name="zoom" value="1" />
      <suscan:field name="rangeLock" value="false" />
      <suscan:field name="timeSpan" value="0" />
    </suscan:object>
    <suscan:object name="audioPanel" class="AudioPanelConfig">
      <suscan:field name="enabled" value="true" />
      <suscan:field name="demod" value="AM" />
      <suscan:field name="rate" value="44100" />
      <suscan:field name="cutOff" value="7350.000000" />
      <suscan:field name="volume" value="77.000000" />
    </suscan:object>
    <suscan:object name="inspectorPanel" class="InspectorPanelConfig">
      <suscan:field name="inspectorClass" value="psk" />
      <suscan:field name="precise" value="false" />
    </suscan:object>
  </suscan:object>
</suscan:serialization>