BatchDrake/SigDigger #99: SigDigger-0.1.0-x86_64.dmg - Unable to add HackRF

PeterDragonTM, 2020-04-25:
When trying to add a HackRF by going to "view --> Devices --> Refresh" in the Capture Devices screen. The SDR HackRF One does not appear in the Device ID list. 

Shut down software, restarted MacOS 10.15.4, no change. 

Opened both CubicSDR and GQRX on MacOS, and the HackRF performs as expected. 

Ia the HackRF supported on the MacOS 0.1.0v?

BatchDrake, 2020-04-25:
This is actually a well-known issue that I plan to fix in 0.1.1. The problem is that I didn't bundle SoapySDR modules in the .dmg and therefore it will look for system-wide installed SoapySDR modules in `/usr/local/lib/SoapySDR/modules0.8`.

I'm marking this as UX issue for traceability. Thanks for reporting!

PeterDragonTM, 2020-04-25:
Sweet thanks!

smdjeff, 2020-07-16:
make sure you grab this too.. 
https://github.com/pothosware/SoapySDR/commit/a8df1c575ff7345fe72050c18ede8c545e255c46

PeterDragonTM, 2020-10-31:
Yeah I gave up, I tried compiling SoapySDR, and SoapyHackRF. then tried Brew install with both. Soapy never wanted to see my HackRF. But other applications outside of Soapy had no issues seeing it. 

Soapy says "Piss off"
>% SoapySDRUtil --probe="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=hackrf
Error probing device: SoapySDR::Device::make() no match

HackRF_Info says "Come on in, I have beer and chips!"
>% hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc21809163
Board ID Number: 2 (HackRF One)
Firmware Version: local-4de2759 (API:1.03)
Part ID Number: 0xa000cb3c 0x00644758

PeterDragonTM, 2021-11-30:
Downloaded the latest build, still get the same issue. Is there any way to force MacOS to use the Soapy drivers that comes with your packaged build?

BatchDrake, 2021-11-30:
Hi,

I am currently reproducing this issue in a macbook (that's something!). It turns out a bug in the .dmg was masking this issue all along. I am already looking into it, thanks for reporting!

BatchDrake, 2021-12-01:
Hi,

I've just scheduled a new development release that may fix this. You should be able to download the new .dmg within the hour. See the progress here: https://github.com/BatchDrake/SigDigger/actions/runs/1526430611

Cheers,

BatchDrake, 2021-12-01:
Nope, still failing. The most frustrating thing is that the bug gets affected by the crappy debugging tools of Mac. If I run suscli with dtruss, I get rtlsdr to be detected. If I run it as-is, it doesn't.

PeterDragonTM, 2021-12-01:
I got it working. Ok so I had to completely remove all Soapy modules and then reinstall them after doing a full update of HomeBrew. 

So after removing all of the following SOAPY modules, I did the following:

brew update
brew upgrade
brew tap pothosware/homebrew-pothos
brew update
brew install soapyremote
brew install soapyaudio
brew install soapyrtlsdr
brew install soapyhackrf
brew install soapybladerf
brew install soapyairspy
brew install soapyairspyhf

The reason I had to do this, was because the previous install of SOAPY had the 0.7 folder and links still active, along with the 0.8 bits. So I believe the software saw the earlier version first, which would make sense why it would fail. Thats a complete guess btw, as a reinstall guts everything. But once I reinstalled the above modules, ignore the airspy bits for this issue, it worked. 

-PD

BatchDrake, 2021-12-02:
I am afraid it is _accidentally_ working for you. The .dmg should be able to use the embedded modules out of the box. 

I've just started a new  build including `SoapySDRUtil` inside the .dmg, hope I can debug this soon.

BatchDrake, 2021-12-02:
Hi again,

I finally managed to debug this. There were three different issues happening at a time:
* Brew somehow installs both SoapySDR 0.8.0 and SoapySDR 0.8.1 (this is bad)
* Modules built for SoapySDR 0.8.1 are not ABI-compatible with SoapySDR 0.8.0
* For some reason, brew links 3rd part software to SoapySDR 0.8.0 and modules to SoapySDR 0.8.1
* Again, for some reason, brew links libSoapySDR.0.8.dylib to libSoapySDR.0.8.1.dylib
* SoapySDR's remote support crashes at startup with SoapySDR 0.8.1
* SDRPlay support freezes the process at startup
* 
I found a workaround which consists of fixing the library hell spawned by brew/SoapySDR directly in the GitHub containers and removing Remote and SDRPlay support. A new development release should be available soon.

BatchDrake, 2021-12-02:
The release is ready, check it out: https://github.com/BatchDrake/SigDigger/releases/tag/latest

BatchDrake, 2023-04-12:
I am closing this issue due inactivity. Feel free to reopen it if the problem persists.

closed 2023-04-12 by BatchDrake