BatchDrake/SigDigger #100: Cannot create audio playback object: audio support disabled at compile time

robertofocke, 2020-04-26:
I installed SigDigger on kali linux. Cannot create audio playback object: audio support disabled at compile time.
[image]

BatchDrake, 2020-04-26:
You need to install `libasound2-dev` with `apt` and then build SigDigger (and only SigDigger) from scratch, starting by a `make clean` on SigDigger's source directory.

robertofocke, 2020-04-26:
[image]

BatchDrake, 2020-04-26:
Wow. This is odd. It seems like a soundcard issue. Can you play audio in your Kali at all? Could you show me your virtual soundcard config in VirtualBox?

robertofocke, 2020-04-26:
[image]

robertofocke, 2020-04-26:
sory, used sudo. it was my error

goscickiw, 2020-05-27:
I'm having the exact same issue. I tried installing `libasound2-dev`, then compiling again after `make clean`, and also after getting a completely fresh copy of the repository. I did `sudo make uninstall` every time before installing the new compilation. What else could I try? (I'm compiling from develop)

BatchDrake, 2020-05-27:
@goscickiw what is the exact error? Audio support disabled or Invalid argument? In case it's the first one, could you run:

```
% pkg-config --cflags --libs alsa
```

And paste the output here?

goscickiw, 2020-05-27:
The error is this:
[image]
`pkg-config --cflags --libs alsa` gives me only this:
``
-lasound
``

BatchDrake, 2020-05-27:
@goscickiw there was a bug in `SigDigger.pro`. Pull from SigDigger (`develop`) and repeat the build from `make clean && qmake SigDigger.pro`

goscickiw, 2020-05-27:
It's fixed now, thanks

closed 2020-05-27 by BatchDrake