BatchDrake/SigDigger #103: Required HackRF version?

jarkko-hautakorpi, 2020-05-09:
HackRF is shown in the Devices, but is not shown in selectable devices.
```
[INFO] Opening HackRF One #0 14d463dc0f4358e1...
[INFO] Could not Open HackRF Device
```
I'v installed HackRF host and tools. Blacklisted the hackrf -module, and tried with it. Disabled USB power save.. No luck.
```
>: hackrf_info
Found HackRF board 0:
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1
Part ID Number: 0xa000cb3c 0x006e4f5b
Serial Number: 0x00000000 0x00000000 0x14d..... 0x0f4.....
```

BatchDrake, 2020-05-09:
Hi,

Could you provide what operating system and SoapySDR version are you using? Also, does it work with Gqrx? Can you provide a screenshot of the Device dialog (Ctrl+D)?

Cheers,

mgp25, 2020-05-10:
| **OS** | **Version** |
|--------|---------------|
| MacOS | Catalina 10.15.4 |

Hello,

Not sure if related but I am facing a similar issue with SigDigger, and it is not able to detect my HackRF device. I have tried with GQRX and CubicSDR and both works.

**GQRX:**

[image]

**CubicSDR:**
[image]

Here is the information:

**HackRF Info:**

```
❯ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8287d46cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x005a434b
```

**SoapySDR infp:**

```
❯ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-release
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libHackRFSupport.so   (0.3.3)
Module found: /usr/local/lib/SoapySDR/modules0.7/libIrisSupport.so     (2018.12.0.1)
Module found: /usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so     (19.04.0)
Module found: /usr/local/lib/SoapySDR/modules0.7/libRedPitaya.so       (0.1.1)
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspySupport.so   (0.1.2)
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspyhfSupport.so (0.1.0)
Module found: /usr/local/lib/SoapySDR/modules0.7/libaudioSupport.so    (0.1.1)
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so   (0.5.1)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so   (0.3.0)
Module found: /usr/local/lib/SoapySDR/modules0.7/libuhdSupport.so      (0.3.6)
Available factories... airspy, airspyhf, audio, hackrf, iris, lime, redpitaya, remote, rtlsdr, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

I haven't achieved to make SigDigger list HackRF in the devices:

[image]

Regards,

BatchDrake, 2020-05-10:
Hi @mgp25 

I assume you downloaded the .dmg from the Releases, right? If that's the case, that's a well known issue. The release in the .dmg file has been linked against SoapySDR v0.8 with no modules whatsoever. I am aware this is extremely annoying and the original plan was to build against the SoapySDR version provided by brew. Unfortunately, brew repos were broken at that time and forced me to build the latest SoapySDR version on my own -- which unfortunately does not match the one in brew. The fix for this is scheduled for the next maintenance release (0.1.1)

Still, the SoapySDR 0.8 library inside the bundle will attempt to look for 0.8 modules in your system inside `/usr/local/lib/SoapySDR/modules0.8`. If you manage to build the HackRF's SoapySDR module (version 0.8) in your system and install it, you shoud be able (in theory) to see HackRF devices. 

If you are still in the mood to make SigDigger work with your HackRF in MacOS, an alternative is to build SigDigger from scratch. Assuming that you have all the required toolkits  installed in your system (Qt 5, Xcode, cmake, SoapySDR development files, device modules and so on), you could try the following:

1. Open a terminal and clone the latest SigDigger release with `git clone`
2. `cd` to the `Scripts` folder inside SigDigger's directory
3. Run `./dist-dmg.sh` and cross your fingers. This script is smart enough to build everything with no human intervention, but requires some stuff to be installed first. If you run into trouble running it, just send me an e-mail and we'll try to make it work.

mgp25, 2020-05-10:
Hello @BatchDrake,

Everything went almost flawlessly but indeed worked! There is a little thing that needs to be updated to avoid some issues. Currently, the `./dist-dmg.sh` script clones develop branches of all dependencies, but there is one that needs a little update. Its related to `SuWidgets`:

```
In file included from Histogram.cpp:21:
./SuWidgetsHelpers.h:26:10: fatal error: 'SuWidgets/Version.h' file not found
#include <SuWidgets/Version.h>
         ^~~~~~~~~~~~~~~~~~~~~
```

Updating `./dist-dmg.sh` to make it clone master instead of `develop` branch should do the trick. I am going to send you an email in order to not "spam" this issue. I am facing an error I am not sure what is causing it.

Best regards

closed 2023-04-12 by BatchDrake