BatchDrake/SigDigger #106: Does support drone fpv 5.8ghz video decoding ?

khanfar, 2020-05-13:
does software support ntsc , pal  analog video ? like decoding drone fpv camera ?

BatchDrake, 2020-05-14:
If you are asking whether you can see the black & white frames of an analog video signal (PAL or NTSC, VSB or FM modulated) the answer is yes. With some effort and parameter adjusting, but yes. The same applies to weather faxes like APT and WeFax. If you are asking whether you can use SigDigger as a replacement for an FPV receiver, the answer is no.

The thing is, the software itself does not directly support these signals. It is more like an incidental feature due to one of its main features: the Symbol View paints the demodulated samples as pixels in a rectangle, filled line by line from left to right, top-down. If you set the bits-per-symbol to a sufficiently high number, the samples are painted according to a smooth grayscale that in turn allows you to inspect picture data of analog video signals like NTSC, PAL, WeFax, SSTV and so on.

Have in mind that, although you can extract the frames, it's unlikely you can do this in realtime. VSB-modulated PAL is at least 6 MHz wide, and FM-modulated PAL / NTSC is even more. Also, there's no way (for now) to perform automatic frame/line synchronization or to demodulate the color subcarrier. These are features I have in mind but will require a lot of experimenting on my own before releasing them.

BatchDrake, 2020-05-14:
As an example, let's take the PAL signal uploaded here: https://www.sigidwiki.com/wiki/PAL_Broadcast

This is what SigDigger would let you do:
[image]
[image]

BatchDrake, 2023-04-12:
I am closing this issue as we already have an Analog TV decoder.

closed 2023-04-12 by BatchDrake