BatchDrake/SigDigger #107: Fresh install (OS/X Mojave) detects neither RTLSDR nor HackRF

cjheath, 2020-05-14:
Fresh SigDigger installed on OS/X Mojave 10.14.6 from https://github.com/BatchDrake/SigDigger/releases/download/v0.1.0/SigDigger-0.1.0-x86_64.dmg

Does not detect RTLSDR dongle or HackRF. Both work with Soapy, and both are visible in the USB tree under System Information.app. I've waited, I've hit Refresh, nothing.

No error messages, so no chance of me making progress with diagnosing this.

BatchDrake, 2020-05-14:
Duplicate of https://github.com/BatchDrake/SigDigger/issues/99

cjheath, 2020-05-14:
#99 mentions the HackRF. Is it also the case that RTLSDR will fail to detect?
I have installed soapysdr using "brew" which provides version 0.7.2, not 0.8.

SigDigger should indicate what version it expects, and where it is looking for the install (and perhaps, allow the user to set an option to look elsewhere)

BatchDrake, 2020-05-14:
Yes, in the end it is basically a module path issue, so it will not work unless you install 0.8 modules. This has been giving a lot of headaches to other users too, so I will definitely downgrade to 0.7 in the next maintenance release (0.1.1) and make sure that brew-installed modules work out of the box.

You can find additional information on this issue here: https://github.com/BatchDrake/SigDigger/issues/103#issuecomment-626335832

royldean, 2020-05-14:
This may be a related issue.   I just installed SigDigger after reading about it on RTL-SDR.com.   Ubuntu 18.04.    The software starts, but fails to recognize the RLT-SDR.com V3 dongle.   Only "Dummy Device" is listed regardless of refreshing attempts.   GQRX starts and recognizes the dongle instantly, if that helps (I realize GQRX isn't using Soapy, I think.....).    Any advice?

jeffpc, 2020-05-14:
You can use SoapySDRUtil to see if Soapy sees the device.  If not, then make sure you have the soapy rtlsdr plugin/module installed.

BatchDrake, 2020-05-14:
@royldean are you running SigDigger from the AppImages or is it a custom build? In case you are using AppImage, are you running the full or the lite release? If not, what files do you have under `/usr/lib/SoapySDR`? (or `/usr/lib/x86_64-linux-gnu/SoapySDR`, don't know where Ubuntu stores SoapySDR modules)

royldean, 2020-05-14:
> You can use SoapySDRUtil to see if Soapy sees the device. If not, then make sure you have the soapy rtlsdr plugin/module installed.

Doh!   Didn't know about the rtlsdr plugin.    Built that and now everything runs!

royldean, 2020-05-14:
> @royldean are you running SigDigger from the AppImages or is it a custom build? In case you are using AppImage, are you running the full or the lite release? If not, what files do you have under [...]

This was a custom build.   But JeffPC guessed correctly, I didn't have the rtlsdr soapy pluggin installed.   Built that and SigDigger showed my dongle immediately.    Thanks!

ah1102, 2020-05-24:
Cool program! I found out about her just now. I would like to see a fix for the sdr device definition issue. I downloaded the latest version of your program for macOS, but the hackrf device was not found.

BatchDrake, 2020-05-24:
@ah1102 this is filed and marked for the maintenance release 0.1.1. See #99 and https://github.com/BatchDrake/SigDigger/issues/103#issuecomment-626335832 for a workaround (requires compiling).

PeterDragonTM, 2020-05-24:
> <https://github.com/BatchDrake/SigDigger/issues/99> and #103 (comment)

When do you think the "maintenance release 0.1.1" will be ready?

ah1102, 2020-05-24:
The program worked) I updated SoapySDR 0.6 to 0.8 from the source code. And SoapyHackRF library was not installed https://github.com/pothosware/SoapyHackRF

[image]

BatchDrake, 2020-05-24:
Well, I'm currently preparing for an exam I have on June 30th. I'll be advancing some work until then, mostly minor bug fixing, features, etc. And once the exam's over, I plan to move forward to bundling these modules in the `.dmg` and announce the release. My only concern is the availability of a working macOS. I do have a virtual machine with it, but it's terribly slow.

ah1102, 2020-05-24:
Show them your program, and I think they will give you the highest rating.
I just noticed that the program was supposed to optimize performance. There is a delay in response.

BatchDrake, 2020-05-24:
@ah1102 could you open a separate issue for that, and describe how to reproduce that delay? Thanks!

PeterDragonTM, 2020-08-29:
Any news on when you might be releasing the 0.1.1 maintenance release that
was to include the fix for this?

Regards,
-PD

BatchDrake, 2020-08-30:
Yep,  I decided to release 0.2.0 directly with a whole lot of new features, including builtin SoapySDR modules for macOS. I plan to release it at some point in September, being my only blocker having access to a macOS machine (I tried hiring cloud-based build machines, but they were way too expensive for what they let me do with them). Sorry for the lack of updates so far!

closed 2023-04-12 by BatchDrake