BatchDrake/SigDigger #110: More Features?

OKCarl, 2020-05-17:
Would this software package be more useful if you merged it with the code from TV Sharp?

http://rtl-sdr.ru/category/tvsharp

BatchDrake, 2020-05-18:
Well, one of the things I want to add is generic video sync, and existing implementations are extremely relevant to me. However, I can't find the sources of TVSharp and I don't know about the licensing of this software either. Where could I find it?

OKCarl, 2020-05-18:
Good question. The Russian gentleman who set up SDR#, where I found TV Sharp went commercial. Nowadays he spends all of his time at the AIRSPY web site.  Perhaps if you left a message at his Forum.

BatchDrake, 2020-05-18:
Nah, in that case, I think I'll just go ahead and try to implement it myself. I gave a look at TVSharp and the software is rather simple. It does not perform chroma demodulation and does not even seem to perform gamma correction. I'll keep this issue opened as a feature request for traceability.

ValZapod, 2020-08-17:
> perform chroma demodulation and does not even seem to perform gamma correction

Implementation for that is here https://github.com/dcherednik/analogcolor/blob/d4cbd9f4ee9d1348b1ccf64b310952f5a234fafe/src/libanalog/ntsc.c#L85

BatchDrake, 2023-04-12:
Closing the issue since we already have an analog TV decoder.

closed 2023-04-12 by BatchDrake