BatchDrake/SigDigger #111: Poor ALSA performance on FreeBSD

jeffpc, 2020-05-23:
FreeBSD has an ALSA compatibility library, unfortunately it doesn't seem to perform as well as PortAudio.

For example, SigDigger processing rtlsdr feed at 1.024MS/s takes about 12% of my CPU. Then, when I enable an FM audio preview (44.1k sample rate, 22.05 kHz audio cutoff, ~15kHz bandwidth) the CPU utilization jumps up to:

- ALSA: 88%
- PortAudio: 48%

The ALSA test was using unmodified SigDigger code.  The PortAudio version of SigDigger was build with the following patch:

```diff
diff --git a/SigDigger.pro b/SigDigger.pro
index 31f2a5a..68ff28a 100644
--- a/SigDigger.pro
+++ b/SigDigger.pro
@@ -271,17 +271,17 @@ packagesExist(volk) {
 # which tend to be the faster ones. If they are not available, fallback
 # to PortAudio.
 
-packagesExist(alsa) {
-  PKGCONFIG += alsa
-  SOURCES += Audio/AlsaPlayer.cpp
-  DEFINES += SIGDIGGER_HAVE_ALSA
-} else {
+#packagesExist(alsa) {
+#  PKGCONFIG += alsa
+#  SOURCES += Audio/AlsaPlayer.cpp
+#  DEFINES += SIGDIGGER_HAVE_ALSA
+#} else {
   packagesExist(portaudio-2.0) {
     PKGCONFIG += portaudio-2.0
     SOURCES += Audio/PortAudioPlayer.cpp
     DEFINES += SIGDIGGER_HAVE_PORTAUDIO
   }
-}
+#}
 
 unix: LIBS += -L$$SUWIDGETS_INSTALL_LIBS -lsuwidgets
 
```

I have no idea how other systems behave, but at least of FreeBSD is it a major difference.  So, it would be worth it to either make it an easy to use compile-time option, or even something that can be toggled in the setting dialog.

BatchDrake, 2020-05-24:
It's not surprising. ALSA is rather complex, and userland emulation had to be expensive (although I didn't expect it to be that bad).

The best approach here would be to write an OSS implementation of the GenericAudioPlayer (give it a look, derived classes are extremely simple) and make it the default for FreeBSD at compile time. In the meantime, I'd directly modify the .pro file to fallback to PortAudio in FreeBSD, maybe with something like this (I haven't tested this yet):

```
if(packagesExist(alsa)&!freebsd) {
  PKGCONFIG += alsa
  SOURCES += Audio/AlsaPlayer.cpp
  DEFINES += SIGDIGGER_HAVE_ALSA
} else {
  packagesExist(portaudio-2.0) {
    PKGCONFIG += portaudio-2.0
    SOURCES += Audio/PortAudioPlayer.cpp
    DEFINES += SIGDIGGER_HAVE_PORTAUDIO
  }
}
```

I'm now wondering how it performs when you press the "Hold" button in the inspection window. Internally, it behaves exactly like the audio preview, but without going through the soundcard.

jeffpc, 2020-05-24:
I did more testing.  It turns out there's at least one other place that seems to be chewing up CPU.

I tried every combination of source (rtlsdr vs. file), audio output (alsa vs. portaudio), and demod (FM vs. none) and recorded the CPU utilization by the whole process as well as two "interesting" threads.

All test runs were using 1.024 MS/s, 50.4 kHz inspection bandwidth, and audio preview settings of: FM, 44.1 kHz rate, 22.05 kHz cutoff, 40% volume.

The rtlsdr runs used a RLT-SDR v3 device directly attached via USB.  The file replay runs used a cf32 raw file stored on local SSD (the filesystem can supply the contents at about 550MB/s thanks to caching and prefetching).

I modified SigDigger's `PlaybackWorker::play` to set the thread name to "PlaybackWorker".  (For whatever reason, the FreeBSD Qt build doesn't set the thread names even though the docs make it sound like it should based on the class name.)

```
Audio | Source | Demod | Proc CPU | PW CPU | "other" CPU
------+--------+-------+----------+--------+------------
ALSA  | file   | FM    | 146%     | 40%    | 74%
ALSA  | rtlsdr | FM    | 84%      | 40%    | 16%
ALSA  | file   | none  | 106%     | N/A    | 100%
ALSA  | rtlsdr | none  | 10%      | N/A    | 7%
Port  | file   | FM    | 106%     | 0.1%   | 74%
Port  | rtlsdr | FM    | 45%      | 0.1%   | 15%
Port  | file   | none  | 106%     | N/A    | 100%
Port  | rtlsdr | none  | 11%      | N/A    | 7%
```
(PW = PlaybackWorker)

So, I see two performance problems:

1. ALSA is taking up about 40% of a CPU core vs. PortAudio's 0.1% utilization.
2. Using a file as a source completely saturates one core.  I don't know which thread it is - but the thread seems to go away only if the source is changed.  (I called it "other" in the table.)

Is ALSA really better on Linux?  This (along with your comment about ALSA being complex to implement) makes me wonder if SigDigger should *default* to PortAudio *everywhere*.

The replay of a file chewing up so much CPU is also surprising but appears to be a separate issue.

jeffpc, 2020-05-24:
> I'm now wondering how it performs when you press the "Hold" button in the inspection window. Internally, it behaves exactly like the audio preview, but without going through the soundcard.

I tested this. While running, I pressed and held "Hold" for about 5-10 seconds while eyeballing top's output :)

The short version: the additional processing is in the noise

The long version: Using the same 8 configurations (alsa/port, file/rtlsdr, FM/none), the only configs I actually see any CPU utilization increase for involve rtlsdr + no demod.  It doesn't seem to matter if I use ALSA or PortAudio, the total CPU utilization jumps about +4%.  The remaining 6 configs, seem to have CPU increase so small that I couldn't see it.  So, I'd call it +0%.

jeffpc, 2020-05-24:
Oh, and FWIW, the SigDigger.pro snippet you proposed works on FreeBSD - it uses PortAudio.

BatchDrake, 2020-05-24:
Okay. So, as I imagined, the CPU usage comes from the audio processing part, not the DSP. Regarding your previous questions:

>  Is ALSA really better on Linux? This (along with your comment about ALSA being complex to implement) makes me wonder if SigDigger should default to PortAudio everywhere.

At least in my machine ALSA performs better than PortAudio. This was something I benchmarked back in February before the release and, although I don't have the specific figures here, I remember the difference was important.

> Using a file as a source completely saturates one core. I don't know which thread it is - but the thread seems to go away only if the source is changed. (I called it "other" in the table.)

This means that the throttling algorithm (implemented in `suscan/analyzer/throttle.c`) is failing.  Unfortunately, I won't be able to investigate this one until a few weeks :( Debugging this function is hard because of how sensitive it is to timing, and it may take more than a couple of hours to figure what is going on. But since I cannot reproduce this in any of my computers, I'd say either `clock_gettime` or `nanosleep` are to blame (see `suscan_throttle_get_portion`), potentially because of some weird time precision issue in FreeBSD.

BatchDrake, 2020-05-24:
And regarding the changes in the .pro file, I'm going to keep them. Thanks!

jeffpc, 2020-05-24:
> At least in my machine ALSA performs better than PortAudio. This was something I benchmarked back in February before the release and, although I don't have the specific figures here, I remember the [...]

Alright.  Given how sensitive to what system you seem to be, I'm even more inclined to argue for a runtime config option.  However, because of the .pro file change completely fixing it for me, I pretty much guarantee that I won't be coding this runtime config support. :)

> This means that the throttling algorithm (implemented in `suscan/analyzer/throttle.c`) is failing. Unfortunately, I won't be able to investigate this one until a few weeks :( Debugging this function [...]

No problem - as far as I can tell, everything works fine - it's just chewing up a core.  I'll try to debug it some more.  Given that that is unrelated to audio, I think it's ok to close this issue.  I'll make a new one to keep track of the non-audio problem.

BatchDrake, 2020-05-24:
Works for me. Thanks!

closed 2020-05-24 by BatchDrake