BatchDrake/SigDigger #114: I/Q recording stuck on after source switch

jeffpc, 2020-05-24:
Steps to reproduce:

1. select a real device source (e.g., rtlsdr)
2. run
3. click "Record" to start writing I/Q to a file
4. stop
5. change source to a recorded file (e.g., the one we just created)
6. run

Since the recorder panel is disabled, I'd assume that recording is not possible/intended.  A new recording is made that's at least as large as the one being played back.  (I haven't checked what happens when the playback loops around to the beginning.)

FWIW, sometimes, I wish I could click record/stop while replaying a longer recording to get just the interesting part.  But that's different from what this bug is about - complete duplication of the played back source file.

BatchDrake, 2020-05-24:
Confirmed as a bug. The recording state should be reset after stopping the capture. However, it is true that disabling recording for non-realtime sources is a bother (this is something I realized some days ago, during a thing called Capture-The-Signal). I don't mind changing the behavior whatsoever.

BatchDrake, 2020-07-24:
This one should be fixed as well. I also modified the capture recorder behavior to allow recordings even in non-live sources. Changes in `develop` as usual.

jeffpc, 2020-07-25:
Yep, seems to work - both the stopping of the recording when stopping analysis as well as recording while playing back a file.  Thanks!

closed 2020-07-25 by jeffpc