BatchDrake/SigDigger #115: Audio preview volume affects the audio recording

jeffpc, 2020-05-24:
I'm not sure if this is a bug or a feature, but I was surprised this morning when I realized that the volume control in the audio preview panel not only changes the volume of the audio played through the speakers but also the volume of the audio saved to the file.

jeffpc, 2020-05-24:
FWIW, I use the audio recording feature to get an FM demodulated .wav with NOAA APT signals.  Since APT is on an AM subcarrier inside the FM rf signal, any adjustments to the volume mess up the data.  There's probably a better way to do what I want.

BatchDrake, 2020-05-24:
Never thought of it that way. The volume is being set at Inspector level and the audio recording just takes the output of the Inspector to store it into a .wav file. A way to fix this could be to add a method to `AudioPlayback` named `setVolume` that mimics the behavior of the `audio.volume` inspector parameter and multiply the samples in `AudioPlayback::write` by a scalar derived from the volume setting (see `suscan/analyzer/inspector/impl/audio.c`) . Then, inside `Application.cpp`, we would have to rewrite:

```c++
void
Application::setAudioInspectorParams(
    unsigned int rate,
    SUFLOAT cutOff,
    SUFLOAT volume,
    unsigned int demod);
```

so it does not accept volume anymore and sets `audio.volume` to a fixed value, like 1. And finally, a bit of signaling madness. We would have to:

1. Modify `AudioPanel::onVolumeChanged` to emit a different signal (e.g. `volumeChanged`, because currently all changes in audio inspector parameters are notified via the `changed` signal).
2. Add a slot in `AudioMediator.cpp` to capture the new signal and emit a new signal named `audioVolumeChanged` 
3. Add a slot in `Application.cpp` that captures `audioVolumeChanged` and call `this->playBack->setVolume` in it (note that `this->playBack` will be `nullptr` if audio playback is disabled).

BatchDrake, 2020-07-24:
This should be fixed now. Check the changes in `develop` (SigDigger only).

jeffpc, 2020-07-24:
On Fri, Jul 24, 2020 at 09:50:26 -0700, Gonzalo José Carracedo Carballal wrote:
> This should be fixed now. Check the changes in `develop` (SigDigger only).

Are the audio icons supposed to get installed as well?

jeffpc, 2020-07-24:
Never mind. It looks like the icons are somehow packed into some resource file - none of the icons are installed as separate files.

Anyway, I gave it a test and it's working great. The mute functionality is good too.  Just a small nit, the icon difference when muted/not-muted is very minor so it'll take some getting used to.

BatchDrake, 2020-07-24:
> Just a small nit, the icon difference when muted/not-muted is very minor so it'll take some getting used to.

Agreed. I am waiting for a friend of mine to create a full icon theme for SigDigger, but there's no specific deadline for that yet. In in the meantime, if you find a better pair of 16x16 icons for the muted/unmuted state, feel free to send me a PR.

jeffpc, 2020-07-25:
I think this issue can be closed.  (Obviously, feel free to reopen it if you think there is still more to do.)

closed 2020-07-25 by jeffpc