BatchDrake/SigDigger #116: Mystery thread with 100% CPU utilization on FreeBSD

jeffpc, 2020-05-24:
This is the promised continuation of #111. A quick recap:

All test runs were using 1.024 MS/s, 50.4 kHz inspection bandwidth, and audio preview settings of: FM, 44.1 kHz rate, 22.05 kHz cutoff, 40% volume.

The rtlsdr runs used a RLT-SDR v3 device directly attached via USB. The file replay runs used a cf32 raw file stored on local SSD (the filesystem can supply the contents at about 550MB/s thanks to caching and prefetching).

I modified SigDigger's PlaybackWorker::play to set the thread name to "PlaybackWorker". (For whatever reason, the FreeBSD Qt build doesn't set the thread names even though the docs make it sound like it should based on the class name.)

```
Audio | Source | Demod | Proc CPU | PW CPU | "other" CPU
------+--------+-------+----------+--------+------------
ALSA  | file   | FM    | 146%     | 40%    | 74%
ALSA  | rtlsdr | FM    | 84%      | 40%    | 16%
ALSA  | file   | none  | 106%     | N/A    | 100%
ALSA  | rtlsdr | none  | 10%      | N/A    | 7%
Port  | file   | FM    | 106%     | 0.1%   | 74%
Port  | rtlsdr | FM    | 45%      | 0.1%   | 15%
Port  | file   | none  | 106%     | N/A    | 100%
Port  | rtlsdr | none  | 11%      | N/A    | 7%
```

(PW = PlaybackWorker)

The high CPU utilization because of ALSA is tracked in #111.  This issue is about the 100% CPU utilization when playing back a file.

jeffpc, 2020-05-24:
Ok. I did some debugging.  You were correct, `suscan_throttle_get_portion` is misbehaving.  Here are my notes.

The recording uses 1.024 MS/s.

Let's look at the latency histogram of `suscan_throttle_get_portion`.  It looks like a *lot* of the calls to this function take between 1024 and 2048 nsecs:

```
$ sudo dtrace -p 13621 -n 'pid$target::suscan_throttle_get_portion:entry{self->start=timestamp}' -n 'pid$target::suscan_throttle_get_portion:return/self->start/{@=quantize(timestamp-self->start)}'
dtrace: description 'pid$target::suscan_throttle_get_portion:entry' matched 1 probe
dtrace: description 'pid$target::suscan_throttle_get_portion:return' matched 1 probe
^C

           value  ------------- Distribution ------------- count
             512 |                                         0
            1024 |@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@  692432
            2048 |                                         8707
            4096 |                                         117
            8192 |                                         11
           16384 |                                         5
           32768 |                                         0
           65536 |                                         0
          131072 |                                         7
          262144 |                                         0
```

At this point, I want to look at the values of `avail` and `nsecs`.  The simplest way is to patch the code to call a dummy function that DTrace can probe:

```diff
diff --git a/analyzer/analyzer.c b/analyzer/analyzer.c
index 802e831..9465ff7 100644
--- a/analyzer/analyzer.c
+++ b/analyzer/analyzer.c
@@ -35,6 +35,10 @@
 #include "mq.h"
 #include "msg.h"
 
+void dummy(long avail, long nsecs, long dsecs, long dnsecs)
+{
+}
+
 /************************ Overridable request API ****************************/
 SUPRIVATE void
 suscan_inspector_overridable_request_destroy(
diff --git a/analyzer/throttle.c b/analyzer/throttle.c
index f7d6089..88906fb 100644
--- a/analyzer/throttle.c
+++ b/analyzer/throttle.c
@@ -74,6 +74,9 @@ suscan_throttle_get_portion(suscan_throttle_t *throttle, SUSCOUNT h)
             - throttle->samp_count;
       }
 
+      extern void dummy(long avail, long nsecs, long dsecs, long dnsecs);
+      dummy(avail, nsecs, sub.tv_sec, sub.tv_nsec);
+
       if (avail == 0) {
         /*
          * Stream exhausted. We wait a fraction of the time it would take
```

Now, let's look at the distribution of `avail` and `nsecs`:

```
$ sudo dtrace -p 15029 -n 'pid$target::dummy:entry{@avail=quantize(arg0);@nsecs=quantize(arg1)}' -n 'END{printf("avail:");printa(@avail);printf("nsecs:");printa(@nsecs)}' -q 
^C
avail:

           value  ------------- Distribution ------------- count    
              -1 |                                         0        
               0 |                                         9        
               1 |                                         3        
[... 66 lines cut ...]
```
avail:

           value  ------------- Distribution ------------- count    
               2 |                                         0        
               4 |                                         14       
               8 |@@                                       1902     
              16 |@@@@@                                    5373     
              32 |@@@@@@@@@@@@@                            13692    
              64 |                                         44       
             128 |@@@@@@@@@@@@@@@@@@@                      20987    
             256 |                                         205      
             512 |                                         35       
            1024 |                                         68       
            2048 |                                         137      
            4096 |                                         137      
            8192 |                                         17       
           16384 |                                         33       
           32768 |                                         66       
           65536 |                                         132      
          131072 |                                         248      
          262144 |                                         0        
```

And the thread that used to take 100% CPU now takes 21%.  The 10% threshold I went with is arbitrary.  I don't know what a good number would be.

BatchDrake, 2020-05-24:
This looks promising. My guess is that you just have figured out a hysteresis-based flow control (and I'm a huge fan of hystereses), but I'd like to understand the impact of this change myself first. I'll try to give you an answer ASAP.

jeffpc, 2020-05-24:
No problem.  I can run with the 10% hack locally.  I'll report if I encounter any issues.

jeffpc, 2020-05-25:
Alright. Found a problem with the 10% hack - with it, data isn't getting through fast enough.  For example, my 1.024 MS/s signal is getting processed at about 900 kS/s.

BatchDrake, 2020-05-25:
Just out of curiosity, what happens if you replace `CLOCK_MONOTONIC` by `CLOCK_MONOTONIC_PRECISE` in that file?

jeffpc, 2020-05-25:
I replaced both uses of `CLOCK_MONOTONIC` in throttle.c.

- no change in CPU utilization (IOW, still 100%)
- no change in `avail` value distribution in `suscan_throttle_get_portion`

BatchDrake, 2020-07-24:
I've just rewritten the throttle control from scratch to use a totally different approach. Time is discretized in checkpoints of fixed spacing (10 ms or more) and block size. If a block corresponding to a given checkpoint is consumed ahead of time, throttle control will sleep until the next checkpoint is reached. This experimental feature is in suscan's `feature/experimental-throttle` branch. Give it a try.

jeffpc, 2020-07-24:
I built it and it seems to work quite well.  The cpu usage is manageable - I didn't bother reproducing the exact settings, but we're talking ~35% total CPU utilization when replying a file vs. 100% from before. The buffer seems to hover between 99% and 100% at 2.56MS/s, the waterfall is nice and smooth.  I'll leave it running in the background to see if it runs into any trouble.

I looked at the code, and was `SUSCAN_THROTTLE_LATE_DELAY_NS` really supposed to be 50 seconds?

BatchDrake, 2020-07-24:
In fact it should be just 5 seconds. I added an extra zero by mistake. I'll
fix it ASAP.

jeffpc, 2020-07-24:
On Fri, Jul 24, 2020 at 14:52:50 +0000, Gonzalo José Carracedo Carballal wrote:
> In fact it should be just 5 seconds. I added an extra zero by mistake. I'll
> fix it ASAP.

That was my guess :)

BatchDrake, 2020-07-25:
I've been playing with the new throttle implementation for a while and I didn't notice anything remarkable (apart from being slightly faster even in Linux). I'm happy with the results so far: I've been replaying a 10 Msps file, 65536-bin FFT and 25 fps, with FM audio demod on (CPU usage ~73%) and off (CPU usage ~53%). Unless you report some weird last-minute issue, I will merge this branch into `develop` later today.

jeffpc, 2020-07-25:
On Sat, Jul 25, 2020 at 07:47:58 -0700, Gonzalo José Carracedo Carballal wrote:
> I've been playing with the new throttle implementation for a while and I
> didn't notice anything remarkable (apart from being slightly faster even
> in Linux). I'm happy with the results so far: I've [...]

I played with it a bunch yesterday, and I didn't see any bad behavior.

BatchDrake, 2020-07-25:
Cool, in that case I'll go ahead and merge this right away. I'm closing this issue too. Thanks!

closed 2020-07-25 by BatchDrake