BatchDrake/SigDigger #121: RTL-SDR not detected on Raspberry Pi 4

Pyro-Fox, 2020-06-02:
I'm going to guess it's just my inexperience, but I just can't figure out why SigDigger doesn't see my RTL-SDR v3. It took me a while, but I managed to get GQRX to see the device, so I know it's possible, but I still only get the dummy device in SigDigger. I don't even know what info I can provide to help troubleshoot.

I've tested running as Sudo, and it doesn't help. If someone can just throw ideas my way, I'll be happy to try them right away.

**How I setup the SDR:**
https://ranous.files.wordpress.com/2020/05/rtl-sdr4linux_quickstartguidev20.pdf

**Device as seen by GQRX:**
[image]

**Proof I'm getting a signal:**
[image]

**Device list from SigDigger:**
[image]

BatchDrake, 2020-06-02:
It looks to me that the corresponding SoapySDR's RTL-SDR module is missing. You are running Raspbian, right? How have you installed SoapySDR?

jeffpc, 2020-06-02:
`SoapySDRUtil --info` can be used to print out what parts of Soapy are installed and what devices it sees.

Pyro-Fox, 2020-06-02:
`pi@raspberrypi:~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
`

Pyro-Fox, 2020-06-02:
Eureka! SoapySDR was the issue after all. After installing [SoapyRTLSDR](https://github.com/pothosware/SoapyRTLSDR/wiki) I can see it now. Thanks!

[image]

closed 2020-06-02 by Pyro-Fox