BatchDrake/SigDigger #122: Waterfall frequency scale labels become useless > 1 GHz

jeffpc, 2020-06-02:
Once the frequencies get high enough, the labels don't show enough precision to provide an indication about how wide the view is.  For example, the following is a screenshot of a signal at approximately 1.698GHz.  Note that all the labels say the same thing: "1.698G".  The only indication that this is a rather zoomed in view, is the audio preview overlay on the spectrum and knowing that it is set to 24kHz.

[image]

My initial idea how to solve this is to have major/minor labels.  At least one major one that gives a nice round frequency, and minor labels that show +- delta from the nearest/nearest-less-than-it major label.

Of course, there might be a better way to convey the frequency info than what I came up with in just now.

BatchDrake, 2020-06-03:
I'm leveraging this issue to say that it would be nice to overlay timestamps in the waterfall to get an idea of the timing of certain events. This would be especially useful for long timespan spectrums or captured signals in which we mess with the throttle override.

BatchDrake, 2023-04-12:
This should be fixed now. Sorry for the late answer!