BatchDrake/SigDigger #126: Allow not maximizing on startup

jeffpc, 2020-07-25:
SigDigger maximizes itself on startup.  This is a bit annoying with tiling/tabbing window managers.  It'd be nice if there was a way to disable the auto-maximize behavior.

BatchDrake, 2020-07-25:
This should not be happening whatsoever. Maybe it is attempting to restore
full screen? What is the fullscreen status in ~/.suscan/config/uiconfig.xml?

jeffpc, 2020-07-25:
On Sat, Jul 25, 2020 at 11:21:55 -0700, Gonzalo José Carracedo Carballal wrote:
> This should not be happening whatsoever. Maybe it is attempting to restore
> full screen? What is the fullscreen status in ~/.suscan/config/uiconfig.xml?

$ grep -i full ~/.suscan/config/uiconfig.xml
    <suscan:field name="fullScreen" value="true" />
      <suscan:field name="fullRange" value="false" />

Hmm...

What is responsible for setting that bool?  I wonder if my window manager is
somehow making SigDigger believe that it is fullscreened even though it
isn't quite.

Jeff.

BatchDrake, 2020-07-25:
It is an option in the view menu, clicking on it should disable fullscreen.
F11 should work too.

jeffpc, 2020-07-25:
Ok. I clicked it, it fullscreened, I clicked again and it un-fullscreened. Then quit. Now it is false, and starting again makes it non-fullscreen.

I think what happened was:

1. at some point, I started SigDigger and used the View -> "Full screen" menu option
2. I quit, writing out the config with "fullscreen=true"
3. Since then, I've been using the WM keyboard shortcut exclusively since it is easier and works on all programs.  But the bool was still true and not getting changed.

So, I think it is safe to close this bug as "PEBKAC" unless there is some way that SigDigger could notice such (un)fullscreening and update the bool.

BatchDrake, 2020-07-25:
Fullscreen handling is problematic across systems (if I remember well,
macOS handled this differently). But I'll take into account, thanks!

closed 2020-07-25 by BatchDrake