BatchDrake/SigDigger #127: Audio preview squelch

jeffpc, 2020-07-28:
It'd be nice if there was an audio squelch - to avoid having to listen to the noise when previewing the audio.  I'm not sure if the saved to wav file audio should be with or without squelch, I'm specifically talking about the audio coming out of the speakers.

jedi7, 2021-04-29:
I will a little update this feature request :)
Actual state: 
* there is SQL for AM and FM. For AM is possible to set level in percentage. For FM is level some magic.
* audio recording ignores sql

Nice to have state:
* Level for all modulations can be set in decibels (same as shown on the vertical axis). Ideally with gui support and visible line on the FFT.
* recording will save audio only when sql detects signal.

BatchDrake, 2021-04-29:
My concern is about whether it makes sense to use the same technique for all modulations. Although we all know what squelch is, the are multiple ways to implement it, not all of them equally effective:
* First of all, SigDigger's squelch feature is based on the following principle: squelch should tell you whether a signal is present or not with minimal user intervention. Adjusting the level every time goes against that principle.
* In suppressed-carrier signals (DSB, SSB, any PSK, etc) the typical level-based squelch works just fine. This is because the power is more or less evenly distributed along the channel. The problem is that there are two possible definitions for this level:
  *  either as some absolute power at the channel passband filter output or 
  * the power spectral density of the channel (this is the one SigDigger implements, as the power spectral density is already displayed at the left of the spectrum view). 

Both mechanisms are affected by the channel bandwidth: if we assume that the signal is band-limited and the noise floor is flat, an overly wide channel may trigger false positives in the former (as the total noise power increases with the bandwidth) and false negatives in the latter (as the mean signal density decreases with the bandwidth). In both cases, you need to provide information about the noise floor and how many dBs above the noise floor are considered "signal". 
* SNR-based squelch (like the one used by the autosquelch feature) works well with suppressed carrier signals as well: the channel is sensed during a few seconds in which the noise floor is measured and the channel is considered "active" if a certain power threshold above the noise floor is surpassed. This removes the dependency on the noise floor but keeps the dependency on the bandwidth.
* Level-based squelch is also possible with AM, but we can use a more discriminating approach exploiting the knowledge we have about their carrier: if the moving average of the DC level at the synchronous AM demodulator's output is close to unity, we can assume that the PLL acquired lock and is successfully following the signal carrier. Instead of providing a level, you provide the "relative amount" (i.e. a percentage) of synchronization you want before hearing something.
* FM signals are completely different. Although a level-based approach is possible, you can perform a more discriminating level-less squelch by exploiting the fact that most FM detectors are based on some kind of derivative of the phase of the signal: the power spectral density of their output in the absence of an FM signal (just noise) is a monotonically growing function, and therefore you can tell with a high level of certainty whether there is a signal by comparing the power at the outputs of an LPF and an HPF plugged to the demodulator output. This what is implemented in SigDigger and what is used in physical commercial FM receivers.
* Additionally, using a level-based approach on FM / AM by visually inspecting the spectrum will be extremely misleading, as what you see as the "carrier level" is actually the power spectral density averaged in the corresponding FFT bin, which is determined not only by the carrier power but also by the FFT's RBW. Carriers are very narrow, with a bandwidth close to zero (ideally they should behave as Dirac deltas), and their power is spread along the RBW of the FFT bin they are in. What does this mean? This means that if you increase the FFT size by a factor of 2, the carrier's power spectral density will grow up to 3 dB (to make things worse, you can't even be totally sure as the true carrier width is also affected by the frequency stability of the oscillators of both the transmitter and the receiver, giving certain profile to it with wings outside its central FFT bin).

So, long story short, if we introduce a pure level-based squelch for all audio modulations, even if we assume the noise floor is stable (which does not have to be and, in practice, it will never be) if we don't want to have false positives with some modulations we may end up with false negatives with other modulations and vice-versa. The same problem occurs with pure SNR-based approaches. The question is, do we really want that?

jedi7, 2021-04-30:
Thanks for explanation. So we must find another solution. Because actually on FM happens the squelch is triggered sometime too soon, so it makes cracking even if there is no audio. Maybe we can look how it is done in SDR# ?

BatchDrake, 2021-04-30:
It's a possibility. You can also go to /suscan/inspector/analyzer/impl/audio.c (develop) and see how it's done. Maybe it makes sense to allow the user to control the lpf/hpf threshold using the squelch spinbutton.

BatchDrake, 2023-04-12:
Hi,

I am closing this issue due to inactivity. Feel free to reopen it if you guys know about a way to improve this squelch feature (I am still thinking this lpf/hpf threshold spinbox may make sense).

Cheers,

closed 2023-04-12 by BatchDrake