BatchDrake/SigDigger #128: Volume control should based on dB instead of %

terrancewong, 2020-08-09:


jeffpc, 2020-08-09:
Serious question: why? The volume control is for sound card playback only, and computer volume controls are generally in %.

terrancewong, 2020-08-09:
just like radio signal, human ear is log scaled. Linear scale 90-100% changes almost nothing, but 1-10% is very significant. Also WBFM output level is always very low need some (+10~20dB) gain.
volume range -60~+30 dB is kind of reasonable. and below -60 ~= mute.

BatchDrake, 2020-08-09:
I agree with terrance. Current volume control is already logarithmic and a
dB scale sounds more natural to me.

jeffpc, 2020-08-09:
On Sun, Aug 09, 2020 at 06:37:58 -0700, Gonzalo José Carracedo Carballal wrote:
> I agree with terrance. Current volume control is already logarithmic and a
> dB scale sounds more natural to me.

Fair enough.  I know that my head is still solidly stuck in computer world,
and so % is still a more natural way of thinking for me.  Thank you both for
the explanation.

BatchDrake, 2020-08-10:
Done, changes in `develop`. I leveraged this issue to fix a months-old issue related to NaNs in the audio inspector.

jeffpc, 2020-08-10:
I got a compile error in suscan:

```
[ 19%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/audio.c.o
/home/jeffpc/src/oss/dsp/suscan/analyzer/inspector/impl/audio.c:306:13: error: controlling expression type
      '_Complex float' not compatible with any generic association type
    det_x = SU_VALID(x[i]) ? x[i] : 0;
            ^~~~~~~~~~~~~~
/home/jeffpc/bin/dsp/include/sigutils/sigutils/types.h:128:19: note: expanded from macro 'SU_VALID'
#define SU_VALID  isfinite
                  ^
/usr/include/math.h:111:21: note: expanded from macro 'isfinite'
#define isfinite(x) __fp_type_select(x, __isfinitef, __isfinite, __isfinitel)
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/math.h:82:48: note: expanded from macro '__fp_type_select'
#define __fp_type_select(x, f, d, ld) _Generic((x),                     \
                                               ^~~
1 error generated.
```

`SU_VALID` seems to be always `isfinite` which (according to the FreeBSD manpage) is supposed to take a *real* float:

```
     int
     isfinite(real-floating x);
```

So, I'm thinking SU_VALID might have to check both real and imag parts separately.

BatchDrake, 2020-08-10:
I've just written a complex version of the macro, try again and tell me if it works.

jeffpc, 2020-08-10:
On Mon, Aug 10, 2020 at 11:08:39 -0700, Gonzalo José Carracedo Carballal wrote:
> I've just written a complex version of the macro, try again and tell me if it works.

Yep.  Built fine and runs ok as well.

Just a note: When I started it the first time, it set the volume to +30 dB.
I would expect the first run to start at 0 dB.  I don't know if that was
supposed to happen, but I'm guessing that in my case it was because I had
the volume set to 100% before the upgrade and it "rounded" the 100 to the
new max number of 30.

BatchDrake, 2020-08-10:
Yep. However, the default volume level was 50 (%) and I've just changed it to a more reasonable -6 (dB) which is roughly one quarter of the demodulator's output power (i.e. one half of its amplitude).

jeffpc, 2020-08-13:
As far as I can tell, this works as expected.  Should this issue get resolved now or later?

closed 2020-08-14 by BatchDrake