BatchDrake/SigDigger #129: strip(1)ing binaries doesn't save that much but makes debugging harder

jeffpc, 2020-08-13:
E.g.,

```
-rwxr-xr-x  1 jeffpc  jeffpc   3.2M Aug 12 11:35 SigDigger
-rwxr-xr-x  1 jeffpc  jeffpc   3.5M Aug 12 11:35 SigDigger.no-strip
```

jeffpc, 2020-08-13:
I suppose this is true for the whole collection: sigutils, suscan, suwidgets, and sigdigger

BatchDrake, 2020-08-14:
Are you building in debug or release mode? Because if you want to debug SigDigger, you have to generate the Makefiles like this:

```
% qmake CONFIG+=debug SigDigger.pro
```
The same idea applies to sigutils & suscan (you have to pass `-DCMAKE_BUILD_TYPE=Debug` to cmake, although I think it defaults to Debug if you don't specify anything). Which makes me wonder, how did you get a non-stripped binary that small?

PS: I've just realized that SigDigger's binary was not being stripped in release mode to begin with. After fixing it, the difference is more dramatic. In release mode, after stripping, I get:

```
% du -sh SigDigger    
3,4M    SigDigger
```

While In debug mode, the binary size is:

```
% du -sh SigDigger
34M     SigDigger
```

jeffpc, 2020-08-14:
On Fri, Aug 14, 2020 at 01:47:12 -0700, Gonzalo José Carracedo Carballal wrote:
> Are you building in debug or release mode?

Release.  When I first figured out how to build SigDigger, I stole the
arguments from one of the scripts in the repository.  That used "release" so
I went with it.  Unfortunately, there is a wide range of definitions of
"debug builds".  For some projects it simply adds `-g` or equivalent to the
compiler invocations (which at least for C and C++ is zero runtime cost).
For other projects, it enables a *ton* of addition asserts which
dramatically slow down execution.  In some cases, it even changes the ABI
of some of their interfaces.

> Because if you want to debug
> SigDigger, you have to generate the Makefiles like this:
> 
> ```
> % qmake CONFIG+=debug SigDigger.pro
> ```

Yes and no. :)

There are two types of information that help debugging:

1. the symbol table
2. the full DWARF info

The first is relatively tiny - in this case, it looks like it is about
300kB.  The second is huge for C++ projects but reasonable for C projects -
based on your numbers, SigDigger's debug info is about 30MB.

The `strip` that's run during `make install` gets rid of both.

My "complaint" is about the symbol table being dropped.  I find it useful to
be able to load up a core file even if it was made by symbol-table-only
binary.  The stack trace still helps narrow things down.

And for the record, as much as I'd love to always have full debug info, I
don't think it makes sense to ship 34 MB binaries.

> The same idea applies to sigutils & suscan (you have to pass
> `-DCMAKE_BUILD_TYPE=Debug` to cmake, although I think it defaults to Debug
> if you don't specify anything).

suscan, etc. are all dynamically linked.  So, it doesn't matter how they are
built.

> Which makes me wonder, how did you get a non-stripped binary that small?

This is the debug info vs. symbol table.

The 3.2MB file has neither.

The 3.5MB file has only a symol table.

> PS: I've just realized that SigDigger's binary was not being stripped in
> release mode to begin with. After fixing it, the difference is more
> dramatic.

It is getting stripped as part of `make install` (release):

```
$ make install
...
/usr/local/lib/qt5/bin/qmake -install qinstall -exe SigDigger /home/jeffpc/bin/dsp/bin/SigDigger
strip /home/jeffpc/bin/dsp/bin/SigDigger
```

That's why I have a script that does:

```
cd SigDigger
PKG_CONFIG_PATH=$TARGET_DIR/lib/pkgconfig/ qmake SigDigger.pro \
	"CONFIG += release" \
	SUWIDGETS_PREFIX=$TARGET_DIR \
	PREFIX=$TARGET_DIR
make
make install
cp SigDigger $TARGET_DIR/bin/SigDigger.no-strip
```

The `cp` is specifically to allow me to have a copy of the binary with a
symbol table.  (There is more to the script.  It rebuilds & installs
everything.)

BatchDrake, 2020-08-21:
> Release.  When I first figured out how to build SigDigger, I stole the
> arguments from one of the scripts in the repository.  That used "release" so
> I went with it.  Unfortunately, there is a wide [...]

That was the problem then. Release binaries are stripped on purpose, while debug binaries are expected to keep both the symbol table and the debug info. In these 4 projects, debug and release binaries are ABI compatible on purpose as well.

Assertions are not going to be an issue either (I don't even think there's any, or any debug-specific path), but you may experience a slight degradation in performance as both sigutils and suscan are compiled with -O0 in order to remove compile-time optimizations and ease debugging. 

In any case, I don't see why we would like to have a Release binary with a symbol table, at least from the perspective of the regular user. If we want to debug SigDigger we should be using debug binaries to begin with.

> And for the record, as much as I'd love to always have full debug info, I
> don't think it makes sense to ship 34 MB binaries.

We never ship debug binaries, those are intended for us to fix stuff and make performance issues more evident. AppImages and other binary releases are always built in Release mode using -O3, -ffast-math and all those fancy command line flags.

> suscan, etc. are all dynamically linked.  So, it doesn't matter how they are
> built

Actually it does, but in a restricted sense. An unstripped dynamic library contains a `.symtab` section which includes static functions (with mangled names like `__FUNCTION__.7706`) and extra symbols that cannot be found in the usual `.dynsym` used for dynamic linking. These extra symbols are rather convenient for debugging.

> It is getting stripped as part of `make install` (release):

This seems to be a system-dependant qmake issue, because I had to enforce it in SigDigger.pro as a GCC option. This is what happens in my system when I run make install, both in debug and release modes:

```
% sudo make install
/usr/bin/qmake -install qinstall /home/waldo/Documents/Development/SigDigger/SigDigger.desktop /usr/local/share/applications/SigDigger.desktop
/usr/bin/qmake -install qinstall /home/waldo/Documents/Development/SigDigger/RMSViewer.desktop /usr/local/share/applications/RMSViewer.desktop
/usr/bin/qmake -install qinstall /home/waldo/Documents/Development/SigDigger/icons/SigDigger.png /usr/local/share/icons/hicolor/256x256/apps/SigDigger.png
/usr/bin/qmake -install qinstall -exe SigDigger /usr/local/bin/SigDigger
```

Could you pull everything from `develop` and tell me what happens with `make install` in both modes? We may need to make this behavior explicit across platforms in `SigDigger.pro`

jeffpc, 2020-08-27:
On Thu, Aug 20, 2020 at 17:12:27 -0700, Gonzalo José Carracedo Carballal wrote:
> That was the problem then. Release binaries are stripped on purpose, while
> debug binaries are expected to keep both the symbol table and the debug
> info. In these 4 projects, debug and release [...]

Thanks good to know.

...
> In any case, I don't see why we would like to have a Release binary with a
> symbol table, at least from the perspective of the regular user. If we
> want to debug SigDigger we should be using debug binaries to begin with.

I like core files, but the core file is quite useless without something to
make the debugger understand what's where - be it the symbol table or the
DWARF info.

In general, I want to be a SigDigger user.  I only want to dive into the
code when something is misbehaving or when I want to improve something.
Therefore, using your definitions, I should be running release binaries
until I have to touch the code.  Unfortunately, it isn't always clear how to
reproduce a SIGSEGV.  This is where (useful) core files help save the day.
Or at least they point the direction by providing (symbolic) backtrace.

> We never ship debug binaries, those are intended for us to fix stuff and
> make performance issues more evident. AppImages and other binary releases
> are always built in Release mode using -O3, [...]

Right, I'm *not* saying we should ship debug binaries.  I *am* advocating
for making release builds at least semi-useful when loading them in a
debugger.

That is, I'd like:

1. release build: with symbol table, without DWARF (due to size)
2. debug build: with *both* a symbol table and DWARF

In both cases, I'm ok leaving the optimization flags the way they are.

> > suscan, etc. are all dynamically linked.  So, it doesn't matter how they are
> > built
> 
> Actually it does, but in a restricted sense.

That was a poor choice of words on my behalf.  What I was trying to say,
that whether or not the symbol table and/or DWARF is in no way dependant on
the optimization level.
 
...
> > It is getting stripped as part of `make install` (release):
> 
> This seems to be a system-dependant qmake issue, because I had to enforce
> it in SigDigger.pro as a GCC option.

Hm, interesting.

...
> Could you pull everything from `develop` and tell me what happens with
> `make install` in both modes? We may need to make this behavior explicit
> across platforms in `SigDigger.pro`

Release:

$ make install
make -f Makefile.Release install
/usr/local/lib/qt5/bin/qmake -install qinstall -exe libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0.1.1
strip --strip-unneeded /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0.1.1
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0.1
...

$ make install
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/SigDigger.desktop /home/jeffpc/bin/dsp/share/applications/SigDigger.desktop
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/RMSViewer.desktop /home/jeffpc/bin/dsp/share/applications/RMSViewer.desktop
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/icons/SigDigger.png /home/jeffpc/bin/dsp/share/icons/hicolor/256x256/apps/SigDigger.png
/usr/local/lib/qt5/bin/qmake -install qinstall -exe SigDigger /home/jeffpc/bin/dsp/bin/SigDigger
strip /home/jeffpc/bin/dsp/bin/SigDigger

Debug:

$ make install
make -f Makefile.Debug install
/usr/local/lib/qt5/bin/qmake -install qinstall -exe libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0.1.1
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0
ln -f -s libsuwidgets.so.0.1.1 /home/jeffpc/bin/dsp/lib/libsuwidgets.so.0.1
...

$ make install
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/SigDigger.desktop /home/jeffpc/bin/dsp/share/applications/SigDigger.desktop
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/RMSViewer.desktop /home/jeffpc/bin/dsp/share/applications/RMSViewer.desktop
/usr/local/lib/qt5/bin/qmake -install qinstall /usr/home/jeffpc/src/oss/dsp/SigDigger/icons/SigDigger.png /home/jeffpc/bin/dsp/share/icons/hicolor/256x256/apps/SigDigger.png
/usr/local/lib/qt5/bin/qmake -install qinstall -exe SigDigger /home/jeffpc/bin/dsp/bin/SigDigger