BatchDrake/SigDigger #132: Inspector window left pane visibility issues

jeffpc, 2020-11-09:
(I'm not sure what to call this, so feel free to reword the description.)

Even since the TV demod code went in, the left side panel was essentially impossible to use.  When I open a new inspector, I see:

[image]

If I try to make it bigger (by click & dragging the margin between it and the center pane), *nothing* happens until I get almost to the right side pane at which point the left pane width snaps and completely hides the middle:

[image]

I don't know if somehow SigDigger is getting confused by my tiling/tabbing window manager.

jeffpc, 2020-11-18:
I ended up recording a quick video to demonstrate the problem: http://josefsipek.net/sigdigger-sidepane-snap.mkv

BatchDrake, 2023-04-12:
Hi,

The side panel changed a lot since this issue was opened. Does the current side panel work for you?

Cheers,