BatchDrake/SigDigger #133: Problem with alsa

as3ii, 2020-11-17:
When using SigDigger without PulseAudio installed, a window popup appear reporting: _"Cannot open audio device. Error was: Failed to set sample rate in ALSA player: Invalid Argument"_. No significant output is printed on the terminal

version: SigDigger v0.1.0
Installed from Archlinux's community repository

BatchDrake, 2020-11-17:
Have you tried setting the sample rate in the Audio Preview? What sample rate are you using?

as3ii, 2020-11-18:
> Have you tried setting the sample rate in the Audio Preview? What sample rate are you using?

The problem occur with every audio sampling rate available

BatchDrake, 2020-11-18:
Okay, that's odd. What's the highest sample rate in the list? Also, could you run `aplay -l` and attach the output to the issue?

as3ii, 2020-11-18:
the higher sample rate shown in the dropdown menu is 192kHz. This is the output of `aplay -l`:
```**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC233 Analog [ALC233 Analog]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 1/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 2: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1```

The PCH soundcard works at maximum at 48kHz, wheres the HDMI interface can work at 192kHz (but I don't use it). If it can help, [this](https://github.com/as3ii/dotfiles/blob/master/alsa/asoundrc.pch) is my alsa configuration, the `default` "soundcard" works perfectly with every program

BatchDrake, 2023-04-12:
Hi,

The latest development builds let you tweak the device used for audio output. Would this work for you?

Cheers,

as3ii, 2023-04-28:
I no longer use that audio setup, so I cannot test properly the development build. Sorry

BatchDrake, 2023-04-28:
It's fine, I just wanted to know the status of the unresolved issues.

In any case, I will go ahead and close the issue. Feel free to reopen it if anything related comes up.

closed 2023-04-28 by BatchDrake