BatchDrake/SigDigger #134: segfault when adjusting roll-off of matched filter

as3ii, 2020-11-18:
As per title, when adjusting the roll-off of the matched filter inside the inspector using the slider, the program crash with a segfault, doesn't matter the digital modulation selected. In `journalctl` I found this line before the stacktrace:
`kernel: traps: SigDigger[83855] general protection fault ip:7ffac1ef89c8 sp:7ffa7e5a0778 error:0 in libvolk.so.2.3[7ffac1d17000+321000]`

BatchDrake, 2020-11-18:
Interesting. This is under Arch with the official repository build, right? Could you do the following?

- Make sure `gdb` is installed
- Run `which SigDigger` to print the path of SigDigger's executable (probably `/usr/bin/SigDigger`)
- Run `gdb /usr/bin/SigDigger` in a terminal (use the path obtained in the previous step)
- In gdb's prompt (prefixed by `(gdb)`), type `run` and press Enter
- Reproduce the crash
- gdb should have captured the segfault and printed `(gdb)`, meaning it allows you to type commands again. Type `bt` and press enter.
- You should see a long backtrace now. Attach it to this issue.

as3ii, 2020-11-18:
yes, it's the package from arch repository. Today it was harder to reproduce the segfault, but after some minutes I did it. This is the backtrace:
```
Thread 38 "SigDigger" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff96ffd640 (LWP 97129)]
0x00007ffff60829c8 in ?? () from /usr/lib/libvolk.so.2.3
(gdb) bt
#0  0x00007ffff60829c8 in  () at /usr/lib/libvolk.so.2.3
#1  0x00007ffff7eeb2a8 in su_iir_filt_feed () at /usr/lib/libsigutils.so
#2  0x00007ffff7f1bee8 in suscan_psk_inspector_feed () at /usr/lib/libsuscan.so
#3  0x00007ffff7f22d04 in suscan_inspector_sampler_loop () at /usr/lib/libsuscan.so
#4  0x00007ffff7f22368 in  () at /usr/lib/libsuscan.so
#5  0x00007ffff7f2c119 in  () at /usr/lib/libsuscan.so
#6  0x00007ffff67f23e9 in start_thread () at /usr/lib/libpthread.so.0
#7  0x00007ffff63e1293 in clone () at /usr/lib/libc.so.6
```

closed 2023-04-12 by BatchDrake